HKUST-Aerial-Robotics/VINS-Mono #222: I try to run this project use KITTI，but the result of vio is really bad

xiaonanguniang, 2018-07-31:
KITTI is a dataset which camera and IMU are 10HZ.
I'm wondering whether because of IMU runs too slow,  cause VINS result so terrible.
And if it has possible to use this dataset run VINS and get right result.(Maybe have some dift, but don't keep initalization or reboot )
thanks.

yueshukun, 2018-08-10:
@xiaonanguniang Could you tell how can I run VINS with KITTI dataset, because when I did that, I encountered one question,it's "OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat". Thanks.

xiesc, 2018-08-22:
@xiaonanguniang 
you can use raw data "unsync", which imu topic was published at 100hz

And even though, the result is also bad.
I think the reason is "bad initialization". Cause, the imu excitation was not enougth!\
So i'm trying to overcome this problem by using Lidar odometry results (like LOAM or NDT) for visual scale initialization.

xiaonanguniang, 2018-08-22:
@xiesc thanks

lshGame, 2018-10-17:
I test VINS_Mono in 2011_09_26_drive_0084_unsync，the result is great.
No enough imu excitation is ok, but low frequency of imu and unsync between imu and images cause bad results. 
My method is below:
      1. use unsync data (IMU 100hz) rather than sync data.
      2. remove out-of-oder imu data. Some GNSS data was lost in collection for bad signal, so there are repeated and out-of-oder imu data. 
      3. interpolate imu data to make that every image frame has one imu data with the same timestamp.
      4. convert hour-minute-second timestamp to decimalism timestamp which is used in ros bag.

xiesc, 2018-10-19:
@lshGame 

For the 2 and 3 in your methods, VINS-mono has already considered these condition. So, it seems not necessary. 
And you said your result is great. What's your criterion? Did you using the criterion provided by KITTI?
For my results, the trajectory seems good, but actually with a bad scale.
I had tested it in 2011_09_30_0016,2011_09_30_0020,2011_09_30_0033,2011_09_30_0034 and 2011_10_03_0042.

By the way, could you offer your *.yaml for the test?

Inventoryo, 2019-04-02:
> @xiesc
Did you transform your kitti data into a rosbag?

karnikram, 2019-05-27:
Hi @xiesc @lshGame 

Is there any tool for converting the unsynced dataset into a ros bag file?

Daihangyu-maker, 2020-03-29:
@karnikram @Inventoryo 
I tried to use kitti_to_rosbag, but I just converted part of the kitti data into a rosbag file, including image and imu data. And I don't know why until now.

Daihangyu-maker, 2020-03-29:
> @xiesc
> I recommend you test VINS in 2011_09_26_drive_0084_unsync, where the car moves from a static state (IMU can be excited in some extend) and therefore it's easy for VINS to initialize.
> U can see my *.yaml in the following link.
> https://github.com/lshGame/Test/blob/master/kitti_unsync_config.yaml

hello, i found that your config file link has been invalid, could you please make it again.

hnsywangxin, 2020-03-30:
> Hi @xiesc @lshGame
> 
> Is there any tool for converting the unsynced dataset into a ros bag file?

reference https://zhuanlan.zhihu.com/p/75672946

lshGame, 2020-03-30:
@Daihangyu-maker 
kitti_to_rosbag is for sync data, not for unsync data.
KITTI has very low support for VIO. 
I recommend you test in KAIST if you want to try some autodriving case. 
KAIST supplies a tool which can read and publish data with ros. 
Besides, KAIST has better synchronization.

lshGame, 2020-03-30:
@hnsywangxin 
晨哥牛b

PetWorm, 2020-03-30:
> @hnsywangxin
> 晨哥牛b

书航认错人了

hnsywangxin, 2020-03-30:
> @hnsywangxin
> 晨哥牛b

我是看到你晨哥的知乎,哈哈

lshGame, 2020-03-30:
@PetWorm @hnsywangxin 
晨哥, 没认错, 我只是给老哥说你很牛B.  :)

gauxonz, 2020-08-24:
hope this helps https://github.com/gauxonz/KITTI-interp @xiaonanguniang

hhhcoder, 2021-06-21:
这突如其来的中文

closed 2021-07-08 by xiaonanguniang

willwilling, 2021-08-25:
> KITTI is a dataset which camera and IMU are 10HZ.
> I'm wondering whether because of IMU runs too slow, cause VINS result so terrible.
> And if it has possible to use this dataset run VINS and get right result.(Maybe have some dift, but don't keep initalization or reboot )
> thanks.

请问你已经解决这个问题了吗？ 能不能分享一下你的config文件，十分感谢

willwilling, 2021-08-25:
> 请问你已经解决这个问题了吗？ 能不能分享一下你的config文件，十分感谢

@xiaonanguniang