HKUST-Aerial-Robotics/VINS-Mono #224: Using VINS-Mono with own device

nPellejero, 2018-08-03:
Hi, I would like to use VINS-Mono with own visual inertial sensor, but right now I have only access to camera, and not IMU, cause the system is still under development. 
So I have proper camera topic, but not IMU topic. I'm trying to use IMU topic from euroc rosbag in addition to camera topic from own camera, but I am not getting any estimation (no messages published in topics from vins_estimator) I guess it's because timestamps ? 
I'm just trying to set up and integrate VINS-Mono to my project  before the actual IMU arrives, there is any chance to do so? what is your insight? 

Thanks in advance,

srikar8, 2018-08-08:
@nPellejero 
yes, the timestamps should be equal for camera and imu topic.

nPellejero, 2018-08-08:
Thanks, @srikar8 ,
So there is any chance to do what I need to achieve? Use camera topic from my camera and replace IMU topic from dataset, temporarily?

srikar8, 2018-08-09:
@nPellejero 
As of now what ever I have gone through, it wont work for your case.
camera and imu should be from the same device.

nPellejero, 2018-08-09:
Ok thanks @srikar8 , I guess I will wait for the sensor to be ready.

closed 2018-08-09 by nPellejero

zanazakaryaie, 2020-07-13:
Isn't there any possibility to use VINS without IMU? I want to use it only as a monocular VO.