HKUST-Aerial-Robotics/VINS-Mono #225: Allow output odometry while initialising

Changliu52, 2018-08-15:
Hi @qintonguav,
When flying drones with VINSMono, everything works really nice. However, the annoying part is that I always has to initialise it by holding and moving the drone.

I wonder if there is a way to allow outputting pose results even when initialising?
Okvis does it, despite they also use optimisation approach. Rovio can also do this.

I believe it is very useful to have this feature especially in drone operations.

Any help will be really appreciated.
Thank you in advance.

Changliu52, 2018-08-28:
Any ideas? Or is this even possible with this package?

yangtzey, 2018-09-10:
@Changliu52 what device do you used?  camera with imu OR an imu device and a camera device?
how to calculate the imu and camera? which tools do you use?i use the kalibr but it works very bad.
could you help me?