HKUST-Aerial-Robotics/VINS-Mono #232: Use IMU device and camera device run VINS-Mono always system reboot

yangtzey, 2018-09-10:
I use a IMU device and camera device.
use imu_utils calibration IMU: https://github.com/gaowenliang/imu_utils  
use image_pipeline calibration camera: https://github.com/ros-perception/image_pipeline
use kalibr for IMU and camera: https://github.com/ethz-asl/kalibr

after those change the YAML file like this:
`%YAML:1.0

#common parameters
imu_topic: "/imu"
image_topic: "/usb_cam/image_raw"
output_path: "/home/tony-ws1/output/"

#camera calibration 
model_type: PINHOLE
camera_name: camera
image_width: 640
image_height: 480
distortion_parameters:
   k1: 0.1350009574451397
   k2: -0.3317601654024684
   p1: -0.005471515947915864
   p2: -0.01633279622876363
projection_parameters:
   fx: 776.7094010879397
   fy: 765.9540385748595
   cx: 297.1355470161591
   cy: 198.4996285113285

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 2   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.
                        # 2  Don't know anything about extrinsic parameters. You don't need to give R,T. We will try to calibrate it. Do some rotation movement at beginning. 
#If you choose 0 or 1, you should write down the following matrix.
#Rotation from camera frame to imu frame, imu^R_cam
extrinsicRotation: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ 0.9299097019170228, 0.3578041808362729, 0.08511236371219963,
           -0.36031559642454186, 0.9326980476063431, 0.015716964227955276 ,
           -0.073760539950713, -0.04528266961432213, 0.9962473902494198]

#Translation from camera frame to imu frame, imu^T_cam
extrinsicTranslation: !!opencv-matrix
   rows: 3
   cols: 1
   dt: d
   data: [ 0.017492412445821233, 0.1431863731296545,-0.007057787484508348]
   
#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
equalize: 1             # if image is too dark or light, trun on equalize to find enough features
fisheye: 0              # if using fisheye, trun on it. A circle mask will be loaded to remove edge noisy points

#optimization parameters
max_solver_time: 0.0  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
#acc_n: 6.4665895894132930e-12          # accelerometer measurement noise standard deviation. #0.2
#gyr_n: 2.1916404833994805e-03        # gyroscope measurement noise standard deviation.     #0.05
#acc_w: 9.0201496474044446e-15         # accelerometer bias random work noise standard deviation.  #0.02
#gyr_w: 2.6750364937251753e-05       # gyroscope bias random work noise standard deviation.     #4.0e-5
acc_n: 0.03415687977864292
gyr_n: 0.0021427985960236697
acc_w: 0.0005900888820947295
gyr_w: 4.399362709089755e-05
g_norm: 9.8035     # gravity magnitude

#loop closure parameters
loop_closure: 1                 # start loop closure
load_previous_pose_graph: 0     # load and reuse previous pose graph; load from 'pose_graph_save_path'
fast_relocalization: 0          # useful in real-time and large project
pose_graph_save_path: "/home/yangtze/catkin_ws/save" # save and load path

#unsynchronization parameters
estimate_td: 0                     # online estimate time offset between camera and imu
**td**: -0.0710461650827               # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#rolling shutter parameters
rolling_shutter: 0                  # 0: global shutter camera, 1: rolling shutter camera
rolling_shutter_tr: 0               # unit: s. rolling shutter read out time per frame (from data sheet). 

#visualization parameters
save_image: 1                   # save image in pose graph for visualization prupose; you can close this function by setting 0 
visualize_imu_forward: 0        # output imu forward propogation to achieve low latency and high frequence results
visualize_camera_size: 0.4      # size of camera marker in RVIZ`

**but it works wrong!**
the result showed below:

> `[ WARN] [1536567841.174215496]: gyroscope bias initial calibration 0.0546195 -0.186871  0.531016
[ INFO] [1536567841.175200983]: misalign visual structure with IMU
[ WARN] [1536567841.344754525]: gyroscope bias initial calibration  0.148928  0.119298 -0.247462
[ INFO] [1536567841.364618933]: Initialization finish!
[ INFO] [1536567844.200319393]:  big z translation
[ WARN] [1536567844.200340903]: failure detection!
[ WARN] [1536567844.200560181]: system reboot!
`

Where is the problem?

I wonder if it is allowed to iMU and camera are two devices.  this algorithm is only suitable for cameras with IMU?

If the IMU and camera are two devices, how to get the td in the yaml file?

thank you very much! looking forward to your reply.

flshyi, 2018-09-18:
Hardware synchronization is better。
kalibr can estimate td

Ascrin96, 2019-05-24:
@yangtzey Hello, I have the same problem, but I can't launch VINS with my web camera. Were you able to solve this problem? Please, any idea.

mrr-05, 2020-09-02:
Did you try to write 1 in estimate_td??

zhaoworking, 2020-11-28:
> Did you try to write 1 in estimate_td??

I have the same problem,  will it work if i set the estimate_td to 1?

closed 2021-01-22 by yangtzey