HKUST-Aerial-Robotics/VINS-Mono #233: what the mean of "acc_n 、gyr_n、acc_w、gyr_w "in  the file  euroc_config.yaml?

csroccv, 2018-09-11:
in the file euroc_config.yaml,note like:
acc_n: 0.2          # accelerometer measurement noise standard deviation. #0.2
gyr_n: 0.02         # gyroscope measurement noise standard deviation.     #0.05
acc_w: 0.0002         # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 2.0e-5       # gyroscope bias random work noise standard deviation.     #4.0e-5

but I  feel they can't calculate by allen variance,by imu_utils (the tool calculate allen variance).What exactly does they mean?How to get them?How to calculate  them?

lenayounis, 2019-04-13:
i have the same question , did you find an answer?

SchwartzSkipper, 2019-04-16:
> i have the same question , did you find an answer?

you can checkout the imu model of [Kalibr](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#the-noise-model-parameters-in-kalibr), which is similar with VINS's.