HKUST-Aerial-Robotics/VINS-Mono #245: VINS not even initializing

bledem, 2018-11-15:
Hello, I have a webcam and an IMU calibrated, they all are sent with 200Hz and 30Hz, but the VINS is not even starting or initializing, just showing : Not enough features or parallax; Move device around.
Even if move around the system the message wont disappear.
Any explanations?
Thank you

srikar8, 2018-11-15:
calibrate your webcam and try again.

bledem, 2018-11-15:
Thank you for answering, I used kalibr to calibrate my camera, should I calibrate with an other way ?
I looked for the debug outpute and even if my stream if doing correctly, I have 
[DEBUG] [1542267962.931893505]: Adding feature points 0
[DEBUG] [1542267962.931916926]: input feature: 0
[DEBUG] [1542267962.931969975]: num of feature: 0
for each image callback

mp3242, 2019-01-22:
@bledem did you end up finding a solution? For me feature points are being generated (they show up in the "tracked image" in rviz) but they are all failing RANSAC (in rviz, the points are all blue), even with an outrageously high threshold. So all points are rejected in the feature tracker, before they even get to the vins estimator, leading to the same debug messages you have. Failing RANSAC could mean that the camera calibration is wrong – I'm using kalibr as well. I have to look at it more but I'm curious if anyone has any advice here.