HKUST-Aerial-Robotics/VINS-Mono #250: Vins crushing in the current version

mrsp, 2018-12-06:
Hello 

At the current version (6ae13d48bbfb260e715b778c06233cc8a29bf3d0)  after Initialization VINS crushes. 

I've tried it in fresh install of 16.04 with eigen 3.3.3 and openCV 3.3.1. and the EuroC example. 

 Version  b638c081e109267c5b350f03cee526f8aa6d28af (August 17) works fine. 

Thanks in advance

MRwangmaomao, 2018-12-06:
I have tied it for two days and nights , but always have no trajectory display on rviz gui.
Thanks you a lot @mrsp. I used the Version b638c08 (August 17) works fine too .

Natalie-ZG, 2018-12-07:
You can try to delete the line noted with "###...###" following, it should be alright.

estimator.cpp
                map<double, ImageFrame>::iterator it_0;
                it_0 = all_image_frame.find(t_0);

                ### delete it_0->second.pre_integration; ###

                for (map<double, ImageFrame>::iterator it = all_image_frame.begin(); it != it_0; ++it)
                {
                    delete it->second.pre_integration;
                    it->second.pre_integration = NULL;
                }
                all_image_frame.erase(all_image_frame.begin(), it_0);

mrsp, 2018-12-07:
Confirming it works after deleting         ### delete it_0->second.pre_integration; ###

What about memory leaks? 

Thanks again

bjdigby, 2018-12-10:
I had the same issue. Commenting out the line mentioned above stopped the estimator from crashing. Thanks!

shaozu, 2018-12-13:
Fixed and very sorry for the inconvenience.

closed 2018-12-13 by shaozu

QiruiGu, 2019-12-06:
why I can not find the sentence in estimator.cpp?

QiruiGu, 2019-12-06:
> You can try to delete the line noted with "###...###" following, it should be alright.
>
> estimator.cpp
> map<double, ImageFrame>::iterator it_0;
> it_0 = all_image_frame.find(t_0);
>
> ```
>         ### [...]

Sorry why I can not find the sentence?