HKUST-Aerial-Robotics/VINS-Mono #251: In old version VINS-Mono,I can run it successfully,but In the latest one,I meet a problem as follow.

shi-gan, 2018-12-08:
[image]

I think it may be different with #70 ,but I don't know how to solve it.Thanks in advanced!

jujumo, 2018-12-11:
Same pb here, it disapear when reverting to b638c081e109267c5b350f03cee526f8aa6d28af.

Here is the backtrace : 
```
*** Error in `/root/catkin_ws/devel/lib/vins_estimator/vins_estimator': munmap_chunk(): invalid pointer: 0x00007f25240bcb20 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f25421e17e5]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x1a8)[0x7f25421ee698]
/root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_ZN15IntegrationBaseD2Ev+0x26)[0x4c9f06]
/root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_ZN9Estimator11slideWindowEv+0xd74)[0x4c6f64]
/root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_ZN9Estimator12processImageERKSt3mapIiSt6vectorISt4pairIiN5Eigen6MatrixIdLi7ELi1ELi0ELi7ELi1EEEESaIS6_EESt4lessIiESaIS2_IKiS8_EEERKN8std_msgs7Header_ISaIvEEE+0xd93)[0x4c7f13]
/root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_Z7processv+0x1466)[0x4a49e6]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80)[0x7f2542b0bc80]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f2543db46ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f254227141d]
```

shi-gan, 2018-12-11:
> Same pb here, it disapear when reverting to [b638c08](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/commit/b638c081e109267c5b350f03cee526f8aa6d28af).
> 
> Here is the backtrace :
> 
> ```
> *** Error in `/root/catkin_ws/devel/lib/vins_estimator/vins_estimator': munmap_chunk(): invalid pointer: 0x00007f25240bcb20 ***
> ======= Backtrace: =========
> /lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f25421e17e5]
> /lib/x86_64-linux-gnu/libc.so.6(cfree+0x1a8)[0x7f25421ee698]
> /root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_ZN15IntegrationBaseD2Ev+0x26)[0x4c9f06]
> /root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_ZN9Estimator11slideWindowEv+0xd74)[0x4c6f64]
> /root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_ZN9Estimator12processImageERKSt3mapIiSt6vectorISt4pairIiN5Eigen6MatrixIdLi7ELi1ELi0ELi7ELi1EEEESaIS6_EESt4lessIiESaIS2_IKiS8_EEERKN8std_msgs7Header_ISaIvEEE+0xd93)[0x4c7f13]
> /root/catkin_ws/devel/lib/vins_estimator/vins_estimator(_Z7processv+0x1466)[0x4a49e6]
> /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80)[0x7f2542b0bc80]
> /lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f2543db46ba]
> /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f254227141d]
> ```

I have solved the problem,you can see #250,It may help you.

shaozu, 2018-12-13:
Fixed and very sorry for the inconvenience.

closed 2018-12-13 by shaozu

QiruiGu, 2019-12-06:
Sorry but I met the same question, and I can not fix it by any means...
[image]

QiruiGu, 2019-12-06:
I can not find the sentence mentioned in estimator.cpp ......
[image]

QiruiGu, 2019-12-06:
I can not find the sentence mentioned in estimator.cpp
[image]