HKUST-Aerial-Robotics/VINS-Mono #253: [vins_estimator-3] process has died

xav12358, 2018-12-13:
Hello 
I use on linux 16.04 with ros kinetic. When I run Vins I get that error;

`roslaunch vins_estimator euroc.launch
... logging to /home/xavier/.ros/log/b8ca2398-fec3-11e8-95ae-b083febd5320/roslaunch-xavier-Inspiron-2595.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.10.74:46369/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/xavier/catk...
 * /feature_tracker/vins_folder: /home/xavier/catk...
 * /pose_graph/config_file: /home/xavier/catk...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /vins_estimator/config_file: /home/xavier/catk...
 * /vins_estimator/vins_folder: /home/xavier/catk...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [2605]
ROS_MASTER_URI=http://192.168.10.74:11311

setting /run_id to b8ca2398-fec3-11e8-95ae-b083febd5320
process[rosout-1]: started with pid [2621]
started core service [/rosout]
process[feature_tracker-2]: started with pid [2636]
process[vins_estimator-3]: started with pid [2641]
process[pose_graph-4]: started with pid [2646]
[ INFO] [1544697721.255557417]: init begins
[ INFO] [1544697721.304568702]: Loaded config_file: /home/xavier/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
result path /home/tony-ws1/output//vins_result_no_loop.csv
[ INFO] [1544697721.305522058]: ROW: 480.000000 COL: 752.000000 
[ WARN] [1544697721.305549720]:  fix extrinsic param 
[ INFO] [1544697721.305670363]: Extrinsic_R : 
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
[ INFO] [1544697721.305712780]: Extrinsic_T : 
-0.0216401  -0.064677 0.00981073
[ INFO] [1544697721.305740230]: Synchronized sensors, fix time offset: 0
[ WARN] [1544697721.305783668]: waiting for image and imu...
vocabulary_file/home/xavier/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/home/xavier/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_pattern.yml
no previous pose graph
[ INFO] [1544697770.877200364]: Not enough features or parallax; Move device around
[ INFO] [1544697771.075956576]: Not enough features or parallax; Move device around
[ INFO] [1544697771.177392921]: Not enough features or parallax; Move device around
[ INFO] [1544697771.375488560]: Not enough features or parallax; Move device around
[ INFO] [1544697771.574501220]: Not enough features or parallax; Move device around
[ INFO] [1544697771.674709559]: IMU excitation not enouth!
[ INFO] [1544697771.675051671]: Not enough features or parallax; Move device around
[ INFO] [1544697771.874253201]: IMU excitation not enouth!
[ INFO] [1544697771.874479979]: Not enough features or parallax; Move device around
[ INFO] [1544697772.076641559]: IMU excitation not enouth!
[ INFO] [1544697772.076975425]: Not enough features or parallax; Move device around
[ INFO] [1544697772.176271411]: IMU excitation not enouth!
[ INFO] [1544697772.176511035]: Not enough features or parallax; Move device around
[ INFO] [1544697772.377530073]: IMU excitation not enouth!
[ INFO] [1544697772.377831151]: Not enough features or parallax; Move device around
[ INFO] [1544697772.586546904]: IMU excitation not enouth!
[ INFO] [1544697772.586857110]: Not enough features or parallax; Move device around
[ INFO] [1544697772.675558003]: IMU excitation not enouth!
[ INFO] [1544697772.675934288]: Not enough features or parallax; Move device around
[ INFO] [1544697772.886254610]: IMU excitation not enouth!
[ INFO] [1544697772.886497314]: Not enough features or parallax; Move device around
[ INFO] [1544697773.081911907]: IMU excitation not enouth!
[ INFO] [1544697773.082200291]: Not enough features or parallax; Move device around
[ INFO] [1544697773.174147912]: IMU excitation not enouth!
[ INFO] [1544697773.174391330]: Not enough features or parallax; Move device around
[ INFO] [1544697773.378276878]: IMU excitation not enouth!
[ INFO] [1544697773.378569440]: Not enough features or parallax; Move device around
[ INFO] [1544697773.578138893]: IMU excitation not enouth!
[ INFO] [1544697773.578485469]: Not enough features or parallax; Move device around
[ INFO] [1544697773.676687083]: IMU excitation not enouth!
[ INFO] [1544697773.677123316]: Not enough features or parallax; Move device around
[ INFO] [1544697773.910116754]: IMU excitation not enouth!
[ INFO] [1544697773.910355470]: Not enough features or parallax; Move device around
[ INFO] [1544697774.077733628]: IMU excitation not enouth!
[ INFO] [1544697774.077985410]: Not enough features or parallax; Move device around
[ INFO] [1544697774.177485375]: IMU excitation not enouth!
[ INFO] [1544697774.177793249]: Not enough features or parallax; Move device around
[ INFO] [1544697774.375643724]: IMU excitation not enouth!
[ INFO] [1544697774.376101827]: Not enough features or parallax; Move device around
[ INFO] [1544697774.587309183]: IMU excitation not enouth!
[ INFO] [1544697774.587615489]: Not enough features or parallax; Move device around
[ INFO] [1544697774.686848232]: IMU excitation not enouth!
[ INFO] [1544697774.687110633]: Not enough features or parallax; Move device around
[ INFO] [1544697774.887717810]: IMU excitation not enouth!
[ INFO] [1544697774.888001360]: Not enough features or parallax; Move device around
[ WARN] [1544697775.202518158]: gyroscope bias initial calibration -0.00217371   0.0209758   0.0779102
[vins_estimator-3] process has died [pid 2641, exit code -11, cmd /home/xavier/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/xavier/.ros/log/b8ca2398-fec3-11e8-95ae-b083febd5320/vins_estimator-3.log].
log file: /home/xavier/.ros/log/b8ca2398-fec3-11e8-95ae-b083febd5320/vins_estimator-3*.log
^C[pose_graph-4] killing on exit
[feature_tracker-2] killing on exit
^Cterminate called without an active exception
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

I get the previous log with the code [vins_estimator-3] process has died.
I can't find solution I try with V1_01_easy.bag and  MH_02_easy.bag.

shaozu, 2018-12-13:
Fixed and very sorry for the inconvenience.

closed 2018-12-13 by shaozu

brandnew2017, 2019-03-06:
I met with the same question, so how to solve it? 
Thank you!

ly1997th, 2019-05-07:
Anyone knows why this happens and how to fix it ? I have been looking for sulotions for days.
Thank you!

shaozu, 2019-05-08:
Did you run the latest version of code?

andongchen, 2019-05-15:
@shaozu I run the latest version of code?encounter no privious pose graph?
How do I set up the  /home/chenandong/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml?

chenandong@ubuntu:~/catkin_ws$ roslaunch vins_estimator euroc.launch 
... logging to /home/chenandong/.ros/log/d76cb8fa-76d3-11e9-9c79-ac1f6b80cc90/roslaunch-ubuntu-23521.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:41201/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/chenandong/...
 * /feature_tracker/vins_folder: /home/chenandong/...
 * /pose_graph/config_file: /home/chenandong/...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vins_estimator/config_file: /home/chenandong/...
 * /vins_estimator/vins_folder: /home/chenandong/...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [23534]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d76cb8fa-76d3-11e9-9c79-ac1f6b80cc90
process[rosout-1]: started with pid [23550]
started core service [/rosout]
process[feature_tracker-2]: started with pid [23554]
process[vins_estimator-3]: started with pid [23555]
process[pose_graph-4]: started with pid [23566]
[ INFO] [1557898784.621259788]: init begins
[ INFO] [1557898784.659599223]: Loaded config_file: /home/chenandong/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
result path /home/chenandong/shaozu/output//vins_result_no_loop.csv
[ INFO] [1557898784.660146272]: ROW: 480.000000 COL: 752.000000 
[ WARN] [1557898784.660178556]:  fix extrinsic param 
[ INFO] [1557898784.660403983]: Extrinsic_R : 
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
[ INFO] [1557898784.660468701]: Extrinsic_T : 
-0.0216401  -0.064677 0.00981073
[ INFO] [1557898784.660507312]: Synchronized sensors, fix time offset: 0
[ WARN] [1557898784.660567390]: waiting for image and imu...
vocabulary_file/home/chenandong/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/home/chenandong/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_pattern.yml
no previous pose graph

ly1997th, 2019-05-15:
I fixed my trouble by reinstalling the environment（eigen3.5 ,ceres2.1.0). I met the trouble by an eigen version conflict, which reinstalling could be helpful.

AnfangRobkit, 2019-12-16:
Hi,

I have met the same problem, with ROS Kinetic + Ubuntu 16.04 + the lastest version of the source code:

What's strange is that I could run it without problem serval days before... 

the output:
```shell

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /localhdd/work/hu...
 * /feature_tracker/vins_folder: /localhdd/work/hu...
 * /pose_graph/config_file: /localhdd/work/hu...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vins_estimator/config_file: /localhdd/work/hu...
 * /vins_estimator/vins_folder: /localhdd/work/hu...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [20286]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a9df6252-2001-11ea-afae-901b0e343ae3
process[rosout-1]: started with pid [20299]
started core service [/rosout]
process[feature_tracker-2]: started with pid [20316]
process[vins_estimator-3]: started with pid [20317]
[ INFO] [1576500215.073589763]: init begins
process[pose_graph-4]: started with pid [20329]
[ INFO] [1576500215.081514362]: Loaded config_file: /localhdd/work/huang/catkin_vins/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
result path /home/shaozu/output//vins_result_no_loop.csv
/home/shaozu/output/ not exists, trying to create it 
/home/shaozu not exists, trying to create it 
Failed to create folder /home/shaozu 
[ INFO] [1576500215.083259327]: ROW: 480.000000 COL: 752.000000 
[ WARN] [1576500215.083279943]:  fix extrinsic param 
[ INFO] [1576500215.083381554]: Extrinsic_R : 
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
[ INFO] [1576500215.083399034]: Extrinsic_T : 
-0.0216401  -0.064677 0.00981073
[ INFO] [1576500215.083412472]: Synchronized sensors, fix time offset: 0
[ WARN] [1576500215.083441237]: waiting for image and imu...
vocabulary_file/localhdd/work/huang/catkin_vins/src/VINS-Mono/pose_graph/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/localhdd/work/huang/catkin_vins/src/VINS-Mono/pose_graph/../support_files/brief_pattern.yml
/home/shaozu/output/pose_graph/ not exists, trying to create it 
/home/shaozu/output not exists, trying to create it 
/home/shaozu not exists, trying to create it 
Failed to create folder /home/shaozu 
/home/shaozu/output/ not exists, trying to create it 
/home/shaozu not exists, trying to create it 
Failed to create folder /home/shaozu 
no previous pose graph
```

shaozu, 2019-12-16:
Hi, you need to change the result path in your config file.

AnfangRobkit, 2019-12-16:
> Hi, you need to change the result path in your config file.
First of all, thanks for your reply.

Yes, I have already changed the the path in the config file, e.g. /home/shaozu -> /personal_folder.

But, I did not help totally :(

```shell
SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /localhdd/work/hu...
 * /feature_tracker/vins_folder: /localhdd/work/hu...
 * /pose_graph/config_file: /localhdd/work/hu...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vins_estimator/config_file: /localhdd/work/hu...
 * /vins_estimator/vins_folder: /localhdd/work/hu...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [23313]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8d09faa6-200c-11ea-afae-901b0e343ae3
process[rosout-1]: started with pid [23326]
started core service [/rosout]
process[feature_tracker-2]: started with pid [23343]
process[vins_estimator-3]: started with pid [23344]
[ INFO] [1576504887.582473543]: init begins
process[pose_graph-4]: started with pid [23369]
[ INFO] [1576504887.591372180]: Loaded config_file: /localhdd/work/huang/catkin_vins/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
result path /localhdd/work/huang/catkin_vins/output/vins_result_no_loop.csv
[ INFO] [1576504887.591670322]: ROW: 480.000000 COL: 752.000000 
[ WARN] [1576504887.591688360]:  fix extrinsic param 
[ INFO] [1576504887.591797592]: Extrinsic_R : 
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
[ INFO] [1576504887.591818529]: Extrinsic_T : 
-0.0216401  -0.064677 0.00981073
[ INFO] [1576504887.591833071]: Synchronized sensors, fix time offset: 0
[ WARN] [1576504887.591861401]: waiting for image and imu...
vocabulary_file/localhdd/work/huang/catkin_vins/src/VINS-Mono/pose_graph/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/localhdd/work/huang/catkin_vins/src/VINS-Mono/pose_graph/../support_files/brief_pattern.yml
no previous pose graph
```

It always gives some warning like **no previous pose graph**

Any information is wellcome. :)

shaozu, 2019-12-16:
Hi, if you did not save the previous pose graph, it is reasonable to get such warnings.

AnfangRobkit, 2019-12-16:
> Hi, if you did not save the previous pose graph, it is reasonable to get such warnings.

Aha...
Thx for your reply. I am kind of stupid just now...
Ignore my dummy questions. :)

Anyway thanks for your hints. :)

Pan-long-ming, 2020-10-11:
> Hi, if you did not save the previous pose graph, it is reasonable to get such warnings.

how can i save the previous pose graph

Pan-long-ming, 2020-10-12:
> Hi, if you did not save the previous pose graph, it is reasonable to get such warnings.
> 
> how can i save the previous pose graph

and are there any previous pose graph ，if it is where are they？