HKUST-Aerial-Robotics/VINS-Mono #258: VINS using too much memory

msamogh, 2018-12-20:
I'm trying to run VINS on a video with HD resolution (1280x720), but the memory consumption of the `pose_graph` process quickly shoots up to 2-3 GBs within a few seconds into running it. Eventually, my computer hangs and I'm forced to kill the process / restart my computer.

Is this level of memory consumption expected? Is it due to the resolution I'm using (I did try downgrading it into VGA, but I faced an OpenCV exception)? Does the memory used by pose graph increase indefinitely with the size of the map?

closed 2018-12-23 by msamogh

zgxsin, 2019-12-04:
@msamogh Hi here, i am also experiencing the similar problem. How did you solve it in the end?

msamogh, 2019-12-07:
Just ended up reducing the resolution and ran it on a machine with higher memory. It worked.

foemre, 2022-04-19:
VINS Mono has to store the pose graph info, so there is no way around the increasing memory I think. #383 probably has an answer if you can decipher it.