HKUST-Aerial-Robotics/VINS-Mono #260: How to solve the problem of ROS_INFO(" big translation")?

JiangZhengok, 2018-12-25:
system reboot after [ INFO] [1545707591.886470637]:  big translation.

[ INFO] [1545707591.305493420]: td 0.0008501
[ INFO] [1545707591.387669604]: td 0.0008725
[ INFO] [1545707591.519758169]: td 0.0008521
[ INFO] [1545707591.607997065]: td 0.0008700
[ INFO] [1545707591.681124020]: td 0.0008504
[ INFO] [1545707591.838649830]: td 0.0008498
[ INFO] [1545707591.886470637]:  big translation
[ WARN] [1545707592.266095416]: failure detection!
[ WARN] [1545707592.271069300]: system reboot!

How to solve this problem?? @qintony

JiangZhengok, 2018-12-29:
After successful initialization, will the system restart under the condition of parking and constant speed movement？

liunao123, 2022-07-21:
my test have same problem<big translation ,many time system reboot! occur>.Can you tell me how to solve it. 3Q.