HKUST-Aerial-Robotics/VINS-Mono #261: Dataset for large fisheye

xav12358, 2018-12-29:
Hello,

I want to use the database used in that video:
https://www.youtube.com/watch?v=2zE84HqT0es

The lens used is a large fisheye view (220 °). 

So is there a example on Vins github repository for that dataset ? What keen of calibration tool do you use for that example (OpenCv or Mei calibration)? 

Best regard.

shaozu, 2019-01-15:
Hi, you can try [TUM VI Dataset](https://vision.in.tum.de/data/datasets/visual-inertial-dataset), where the FOV is about 195 degree. Here is the [config](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/config/tum/tum_config.yaml) to run it. We use `KANNALA_BRANDT` camera model as  provided in the website.

closed 2019-01-15 by shaozu

neorobo, 2019-04-16:
I've been looking into VINS implementation of Equidistant and comparing to Kalibr. Is there an easy way to convert between the two? Kalibr provides 4 coefficients, k1-k4, but vins sets k1 to 1 and expects k2-k5. Is k1 set to 1 through some normalization?

Astronirav, 2019-06-03:
The equidistant model is actually Kanalla-Brandt's algorithm. CamOdoCal (which VINS-Mono uses) implemented equidistant model exactly as described in the K-B's paper. However, Kalibr assumes k1 = 1 (so does opencv's fisheye function), which means that the 4 values given by it are actually k2-k5.