HKUST-Aerial-Robotics/VINS-Mono #266: vins_estimator crashes (eigen3.3.3, Ceres 1.14)

antoan, 2019-01-11:
Enviroment: Ubuntu 16.04, ROS Kinetic, OpenCV 3.3.1, Eigen 3.3.3

The estimator node crashes as soon as the Euroc MH_01_easy.bag starts running

How I built Vins Mono:
* I have built and installed Ceres 1.14 with Eigen3.3.3

* Vins-Mono & Eigen 3.3.3:
Edited each package's CmakeLists.txt (feature_tracker, pose_graph, ect ) to hardcode the path to Eigen3.3.3 and commented out the call to FindEigen.cmake:

#find_package(Eigen3)
set(EIGEN3_INCLUDE_DIR "/home/tony/eigen3.3.3")

Any suggestions would be much appreciated.

[VINS-MONO Build log](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/files/2751215/VINS-MONO.Build.log)

> tony@Radiance:~/catkin_ws$

 roslaunch vins_estimator euroc.launch 
... logging to /home/tony/.ros/log/96fa5a56-15e5-11e9-be0a-14dae91b4fe9/roslaunch-Radiance-7132.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Radiance:37613/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/tony/catkin...
 * /feature_tracker/vins_folder: /home/tony/catkin...
 * /pose_graph/config_file: /home/tony/catkin...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vins_estimator/config_file: /home/tony/catkin...
 * /vins_estimator/vins_folder: /home/tony/catkin...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [7145]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 96fa5a56-15e5-11e9-be0a-14dae91b4fe9
process[rosout-1]: started with pid [7158]
started core service [/rosout]
process[feature_tracker-2]: started with pid [7161]
process[vins_estimator-3]: started with pid [7162]
process[pose_graph-4]: started with pid [7163]
[ INFO] [1547241149.111315595]: init begins
[ INFO] [1547241149.118199818]: Loaded config_file: /home/tony/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
result path /home/tony-ws1/output//vins_result_no_loop.csv
[ INFO] [1547241149.213688267]: ROW: 480.000000 COL: 752.000000 
[ WARN] [1547241149.213748686]:  fix extrinsic param 
[ INFO] [1547241149.326545817]: Extrinsic_R : 
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
[ INFO] [1547241149.326597393]: Extrinsic_T : 
-0.0216401  -0.064677 0.00981073
[ INFO] [1547241149.326631479]: Synchronized sensors, fix time offset: 0
[ WARN] [1547241149.326704016]: waiting for image and imu...
vocabulary_file/home/tony/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/home/tony/catkin_ws/src/VINS-Mono/pose_graph/../support_files/brief_pattern.yml
no previous pose graph
[ WARN] [1547241178.943072436]: gyroscope bias initial calibration -0.00193261   0.0482494   0.0797451
[vins_estimator-3] process has died [pid 7162, exit code -11, cmd /home/tony/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/tony/.ros/log/96fa5a56-15e5-11e9-be0a-14dae91b4fe9/vins_estimator-3.log].
log file: /home/tony/.ros/log/96fa5a56-15e5-11e9-be0a-14dae91b4fe9/vins_estimator-3*.log

shaozu, 2019-01-13:
Hi, antoan, are you using the latest version of code? If no, please update the code. And if yes, it seems like a memory crash problem, then you can use [backward](https://github.com/bombela/backward-cpp) to get detailed infomation.

zhuoyueljl, 2019-01-14:
You can try to delete the line noted with "###...###" following, it should be alright.

estimator.cpp
map<double, ImageFrame>::iterator it_0;
it_0 = all_image_frame.find(t_0);

        ### delete it_0->second.pre_integration; ###

        for (map<double, ImageFrame>::iterator it = all_image_frame.begin(); it != it_0; ++it)
        {
            delete it->second.pre_integration;
            it->second.pre_integration = NULL;
        }
        all_image_frame.erase(all_image_frame.begin(), it_0);

closed 2019-01-15 by shaozu