HKUST-Aerial-Robotics/VINS-Mono #270: Doubts about camera-imu temporal calibration

odinhr, 2019-02-26:
Hi all!

I want to know what can I do for launching the temporal calibration which is mentioned in the IROS 2018 paper. Every time I've launched the VINS-mono, it returns me "imu message in disorder". Any help is welcome.

Thank you in advance.

mohradza, 2019-02-26:
I noticed that I get this message when my IMU messages are coming in faster than 100hz. After reducing my IMU stream to < 100Hz, I no longer get this message.

gakkiHp, 2019-03-11:
@mohradza 
Hi，I want to know how to reduce the IMU rate?

odinhr, 2019-03-12:
Hi @gakkiHp,
it depends on your setup. Can you give us some information about that?

Regards.

gakkiHp, 2019-03-12:
@odinhr 
I am using the android phone to collect and process the data. I would like to ask if I can directly reduce the imu speed on the vins-mono source code

kharyal, 2020-08-18:
https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/0d280936e441ebb782bf8855d86e13999a22da63/vins_estimator/src/estimator_node.cpp#L142

Found only one line printing this warning. My IMU data is in order and I tried playing with different frequencies for my IMU but it still didn't help.

Best.

zhaoworking, 2020-11-20:
> I noticed that I get this message when my IMU messages are coming in faster than 100hz. After reducing my IMU stream to < 100Hz, I no longer get this message.

Can u tell me how to do that, i have encountered this knotty problem.

kharyal, 2020-11-20:
> > I noticed that I get this message when my IMU messages are coming in faster than 100hz. After reducing my IMU stream to < 100Hz, I no longer get this message.
> 
> Can u tell me how to do that, i have encountered this knotty problem.

I have noticed that as long as your IMU data is in order, the warning doesn't hurt the localisation. It sure does hurt your data rate though.

zhaoworking, 2020-11-20:
> I have noticed that as long as your IMU data is in order, the warning doesn't hurt the localisation. It sure does hurt your data rate though.

I want to know with this WARN , whether the VINS could work finely?

kharyal, 2020-11-20:
Yes, I have seen it work fine even after throwing this warning. Not sure about every possibility, though. You can check the timestamps of your IMU messages and see if they are in order.

Regards.

zhaoworking, 2020-11-20:
> Yes, I have seen it work fine even after throwing this warning. Not sure about every possibility, though. You can check the timestamps of your IMU messages and see if they are in order.
> 
> Regards.

I find the timestamps of the `imu.csv` are right , but when I creat the `rosbag `, and feed it to the Vins , the terminal displays WARN  as well as `INFO imu order : 79484.300000 last_imu_t: 79484.300000` , which are surprisingly same. Why could it happen?

Leerw, 2022-07-27:
> I find the timestamps of the `imu.csv` are right , but when I creat the `rosbag `, and feed it to the Vins , the terminal displays WARN as well as `INFO imu order : 79484.300000 last_imu_t: [...]

I have the same issue with you, the current imu timestamp is the same as the `last_imu_t`, have you fixed that?

yuanshangwu, 2022-10-06:
> 

I also have the same issue with you, have you fixed that?

yuanshangwu, 2022-10-06:
> I have the same issue with you, the current imu timestamp is the same as the `last_imu_t`, have you fixed that?

I also have the same issue with you, have you fixed that?