HKUST-Aerial-Robotics/VINS-Mono #272: How to merge the map?

ran5515, 2019-03-08:
Dear Authors,

How do you merge maps? I have read your [paper](https://arxiv.org/abs/1803.01549) and source code, but can't find the answer. Could you give me some hints about how you do it? Also, while merging, do you assume different sequences have the same start point?

Thanks!

shaozu, 2019-03-08:
Hi, ran5515. Map merging is done by closing loop between the current pose graph and loaded pose graph, and you can read [pose graph part of our code](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/tree/master/pose_graph/src)  if you want more details. BTW, we do not make the assumption that different sequences have the same start point.

closed 2019-03-08 by shaozu

ran5515, 2019-03-08:
@shaozu I see, thank you! I am still looking through it, and find KeyFrame::findConnection function maybe pretty important. 

(I will update my understanding here in case someone else has same question with me.)