HKUST-Aerial-Robotics/VINS-Mono #278: Problems getting VINS-Mono running (wrong calibration?)

KR51K, 2019-03-25:
Hi,

I'm stucking on getting VINS-Mono running on my PC. So here is something about my setup/configuration, maybe you see the/an error of my procedure:
I have a small camera (320x256px, running with 9Hz) and a low cost IMU (10 DOF from DFRobot with an ADXL345 Acc. and an ITG3200 Gyro). The IMU is attached on the bottom of the camera. I have used imu-tools for getting noise and bias of the IMU and kalibr with time-offset calibration flag for  camera-IMU calibration. I have written the result from the yaml file in the config file for VINS-Mono with the upper left 3x3 matrix from the 4x4 Matrix T_ci (imu0 to cam0) as extrinsicRotation and the last column as extrinsicTranslation, and also the td (-0.0009637).
But when I'm running VINS-Mono I get several errors like 
`misalign visual structure with IMU`
or
`Initialization finished!`
`td 0.207015`
`big IMU gyr bias estimation 4.486076`
`failure detection!`
`system reboot!`

Any glue about a solution? I'm sitting on this for several days now but can't handle it...

shaozu, 2019-03-25:
Hi, @KR51K. Actually the performance of VINS-Mono depends a lot on the device you use. Usually it won't work with two arbitrary low-cost unsynchronized camera and IMU. The parameter `td` means the fixed time offset between camera and IMU in synchronized setting, while in an unsync case, this parameter may be time varying and depend on your OS load, thus be meaningless. So you are highly recommended to pick up your device according to section 5.7 of `README.md`. Also, the IMU noise used in VINS-Mono is in discrete case, so make sure to convert it if you obtained noise in continuous case.

KR51K, 2019-03-25:
Hello, thanks for your response! I have already tried  with setting the td to 1, and also to let VINS Mono to estimate the parameters for imu to cam orientation, but this does not help either...

KR51K, 2019-03-26:
Oh man,
standing under the shower I suddenly realized the problem: I had some trouble with the IMU at the beginning and I have read some posts about similar problems. In one post there was mentioned:

> pay attention to your input data's units
1)acc:m/s^2
2)gyo:rad/s

and

> Check you IMU data.
the units should be what @mlyarthur said.
The value is around 9.8

So I had the problem with the accelerometer and changed this, and kept 2) in mind. But only this. So my IMU is posting it's data in °/s not rad/s *facepalm*.

To changing this did it (sometimes it is so easy if you are doing it right...). The tracking is still bad, but now I have a good point to start (try to synchronize cam & iMU, check noise,...)

Thanks for the rapid assistance and best regards!

shaozu, 2019-03-26:
Glad to hear that and good luck!

closed 2019-03-26 by shaozu

sheroy666, 2019-04-02:
> Oh man,
> standing under the shower I suddenly realized the problem: I had some trouble with the IMU at the beginning and I have read some posts about similar problems. In one post there was [...]

Could you tell me how to change the acce data? My acce data needs to multiply 9.8, but I don't know how to do it. Thank you!