HKUST-Aerial-Robotics/VINS-Mono #280: Camera position is moving fast while robot is only tilted and rotated

KR51K, 2019-03-28:
Hi, 

I'm wondering why the camera position is is drifting (sometimes very fast) even if the robot is only tilted and rotated (unfortunately cam and imu are not synronized). While visual feature detection is really stable I see that _position_ in the _imu_propagate_-topic seems to be responsible for that.  So, how could I change this? Is this because of my low-cost (noisy) IMU?

shaozu, 2019-03-30:
Hi, have you checked the IMU noise? is it in discrete time or in continuous time?

KR51K, 2019-03-30:
Hi, i have used imu_tools. But with this settings tracking is even worse than with settings Front euroc yaml file...

shaozu, 2019-03-30:
Hi, may I ask which `imu_tools` are you using?

KR51K, 2019-03-30:
Sorry, I ment imu_utils from GitHub (https://github.com/gaowenliang/imu_utils).

shaozu, 2019-03-30:
@KR51K. Try divide your acc_n and gyr_n by square root of your imu sample time, and multiply your acc_w and gyr_w by square root of your imu sample time.

KR51K, 2019-03-31:
Hi,
thank you so very much, this significantly improves my tracking!

closed 2019-03-31 by KR51K

ns15417, 2019-10-30:
@KR51K hi, may i ask where did you multiply the sqaure root?  should this modification be added inside the code or in the .yaml file when i run the code ? or is it used for the calibration for IMU?

ns15417, 2019-10-30:
@KR51K hi, sorry about the last question, I saw the input configuration file of VINS and saw those params, so I think maybe i should multiply the square root in .yaml file. Right?
I got another question about ORBVI, have you run this project? why does learnORBVI do not need the IMU configuration params? cuz I calibrated IMU but i didnt see any place it use the calibration parameters of IMU, like acc_n, acc_w， even no need to input those calibration data in config .yaml file , do you have any idea about this?

Gatsby23, 2023-04-12:
> @KR51K. Try divide your acc_n and gyr_n by square root of your imu sample time, and multiply your acc_w and gyr_w by square root of your imu sample time.

Sorry to trouble you with this simple question: How can I determine whether the IMU used in this code is discrete or continuous? My previous understanding was that the continuous form of IMU is only used for theoretical derivation, and in practical use, it needs to be converted into a discrete form to write code. However, I saw in the discussion about ORB-SLAM3 that it seems to use the imu under the continuous format.

yeoeun98, 2024-12-13:
> @KR51K. Try divide your acc_n and gyr_n by square root of your imu sample time, and multiply your acc_w and gyr_w by square root of your imu sample time.

Is the unit of imu sample time in seconds?