HKUST-Aerial-Robotics/VINS-Mono #282: When I applied my own data , I met this prolem.

hjr553199215, 2019-03-31:
OpenCV Error: Assertion failed (_mask.empty() || (_mask.type() == (((0) & ((1 << 3) - 1)) + (((1)-1) << 3)) && _mask.sameSize(_image))) in goodFeaturesToTrack, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgproc/src/featureselect.cpp, line 366
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgproc/src/featureselect.cpp:366: error: (-215) _mask.empty() || (_mask.type() == (((0) & ((1 << 3) - 1)) + (((1)-1) << 3)) && _mask.sameSize(_image)) in function goodFeaturesToTrack

[feature_tracker-2] process has died [pid 3247, exit code -6, cmd /home/hujiarui/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/hujiarui/.ros/log/9b635d3e-5377-11e9-9bf0-48a47263891d/feature_tracker-2.log].
log file: /home/hujiarui/.ros/log/9b635d3e-5377-11e9-9bf0-48a47263891d/feature_tracker-2*.log

hjr553199215, 2019-03-31:
imu:100hz   camera:20hz

shaozu, 2019-03-31:
Seems you are using fisheye camera but did not provide any mask image for that.

hjr553199215, 2019-03-31:
I am not sure if the problem happens in the function "goodFeaturesToTrack()" or the fisheye camera.

hjr553199215, 2019-03-31:
Because the "mask" appears both in "fisheye camera" and "goodFeaturesToTrack()" ,so I am confused about the error message.

The-Forgotton-One, 2019-04-11:
Maybe your camera resolution is different form configuration file.
You can check euroc_config.yaml to make sure your camera resolution is right.

G-KUMAR, 2019-05-05:
[image]
 Some how my **error got solved** by disabling fisheye mask("fisheye: 0" in config.yaml file) though I am using a fisheye lens. Don't know the reason yet. The mask dimensions were consistent.

zhuoyueljl, 2019-05-21:
check resolution of image, the problem will be solved!

wallace-lai, 2022-05-11:
您好！您的邮件我已经收到了，我将尽快回复，谢谢！

r-eality-eng, 2023-09-19:
我也遇到这个问题，有人解决了吗