HKUST-Aerial-Robotics/VINS-Mono #285: Solve PnP in Initialization

huanhexiao, 2019-04-22:
in the function bool Estimator::initialStructure(), line 351 of eistimator.cpp are: 
***
frame_it->second.R = R_pnp * RIC[0].transpose();
frame_it->second.T = T_pnp;
***
Here (R_pnp,T_pnp) is the camera's pose just retracted from cv::solvePnP, right? 
i think  "frame_it->second.R = R_pnp * RIC[0].transpose();" transforms the camera's roatation to IMU's. But nothing happens to T_pnp, why?
===========================
Same as the code in line 297:
***
frame_it->second.R = Q[i].toRotationMatrix() * RIC[0].transpose();
frame_it->second.T = T[i];
***
help please, thanks in advance.

shaozu, 2019-04-24:
Because by that time we have not recover the scale, so nothing can be done to T_pnp. Later in line 405, after initialization, we are able to compute the translation of body frame.

huanhexiao, 2019-04-24:
> Because by that time we have not recover the scale, so nothing can be done to T_pnp. Later in line 405, after initialization, we are able to compute the translation of body frame.

Ok, got it. Thanks

closed 2019-04-24 by shaozu