HKUST-Aerial-Robotics/VINS-Mono #289: It seems that there is a programming mistake.

jayoungo, 2019-05-18:
In line 65 of "feature_tracker/src/feature_tracker_node.cpp":
```
    if (img_msg->encoding == "8UC1")
    {
        sensor_msgs::Image img;
        img.header = img_msg->header;
        img.height = img_msg->height;
        img.width = img_msg->width;
        img.is_bigendian = img_msg->is_bigendian;
        img.step = img_msg->step;
        img.data = img_msg->data;
        img.encoding = "mono8";
        ptr = cv_bridge::toCvCopy(img, sensor_msgs::image_encodings::MONO8);
    }
    else
        ptr = cv_bridge::toCvCopy(img_msg, sensor_msgs::image_encodings::MONO8);
```
Data conversion is needed when the type is not "mono8", so the judgement condition should be "!=" rather than "=="? 
Or is my understnding wrong? Thank you!

shaozu, 2019-05-22:
Hi, we find that when fetch images from camera, some drives give `8UC1` as the image encoding, but it's not a valid option in function [`cv_bridge::toCvCopy`](http://docs.ros.org/diamondback/api/cv_bridge/html/c++/namespacecv__bridge.html#acbf2da402f4d3e505613e95b5a2aed35). That's why we need to explicitly make it `mono8` if the encoding is `8UC1`.

jayoungo, 2019-05-22:
@shaozu I get it, Thank you!

closed 2019-05-22 by jayoungo