HKUST-Aerial-Robotics/VINS-Mono #290: Can the device be initialized successfully if it has an initial speed ?

zjtde1990, 2019-05-22:
I am wondering why the initial speed of each pre-integration is 0. If the device has a big initial speed, then v_bk_bk isn't equal to 0. Does this has no effect on the results?

shaozu, 2019-05-22:
Hi, actually velocity has been initialized in the initialization phase, and the pre-integration term (i.e. alpha in the paper) is not the position.

zjtde1990, 2019-05-28:
Sorry, I didn't describe my question clearly.  
My question is :  If the divice is mounted on the car, eg. the car is driving outdoors at a speed of 30 km/h, then I started the program. At this moment, the program starts to initialize. But the initial value of speed state is 0, and in pre-integration the delta_v is 0 (delta_v{Eigen::Vector3d::Zero()}).  In this case, are the scale and velocities correct after initialization?

shaozu, 2019-05-28:
Hi, the initialization will work fine even the initial speed is not 0. The variable `delta_v` in pre-integration is not exactly the velocity, instead it's a pre-integration constraint showed in the paper as `beta` (see formulation 3). So `delta_v` has nothing to do with the initial velocity. As for the actual initial speed of the object, we will initialize that at the end of the initialization.

zjtde1990, 2019-05-29:
Yes, I know what you mean. Thank you very much! 
But I really don't know why my results are wrong. Can you give me some suggestions?
[image]
cT is the translation vector from ck to c_ref, which the camera frames are in sliding window; scale and Vs are the results after initialization. But, in fact, the speed of the car is over 20km/h. I observe the translation vectors, but the trend is correct.  I tried many times in kinds of situations, but the scale result and the velocities were both wrong. I really dont't know the reasons. My device is equipt with a monocular camera and a hardware-synchronized IMU, which is mounted on a car. 
I'm looking forward to your reply. Thanks a lot~

shaozu, 2019-05-30:
I think the IMU excitation during the initialization may not enough. Actually such near-planar motion is really a challenging case for VINS-Mono.

zjtde1990, 2019-05-31:
Oh, you mean the reason for this is that motion degration makes the states unobservable. Maybe I understand something. Thank you very much!
And do you have any suggestions for solutions to the VIO which is used for near-planar motion vehicle applications? Or do you know any resources on that? Thanks~

shaozu, 2019-05-31:
You can try with [VINS-Fusion](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion).

1ADZX, 2021-07-24:
> 你好，其实velocity在初始化阶段就已经初始化好了，预积分项（即论文中的alpha）不是位置。

Hello, I can understand that the alpha here is not the position, but the relative displacement of IMU from K to K + 1, but suppose I calculate the relative displacement from frame 0 to frame 1 of IMU data,  The initial value of delta_V will certainly have an impact on my calculation here, and I found that there is no other parts of the code Where delta_V is assigned other values, except that the initial value of this value is 0

1ADZX, 2021-07-24:
> 哦，你是说这样做的原因是运动退化使状态不可观察。也许我明白了什么。非常感谢你！
> 您对用于近平面运动车辆应用的 VIO 的解决方案有什么建议吗？或者你知道这方面的任何资源吗？谢谢~

你和我的最近的疑问基本一致，然后看了你们上述的回复，对我启发很大，的确预积分后一帧相对前一帧的位姿变换，不能当作是在世界坐标系下的位姿变换，需要再进行额外的计算才能得出世界坐标系下的变换。我自己仿真的IMU数据，也是初值不为0，但刚又结合公式和对应的预积分计算数据，的确是这里delta_V初值为0，算出的预积分alpha，beta都是相对前一帧的，因此数值是正确的，然后再结合公式转换到世界坐标系下，就可以得到真实的位姿变换了。因此，源代码是正确的