HKUST-Aerial-Robotics/VINS-Mono #296: Large drift when using T265 with IMU values given

yx0123, 2019-06-10:
I am using the Intel Realsense T265 which has global shutter camera and hardware synchronization. When I run VINS Mono with using the default IMU parameters for the example dataset, there is no huge drift but the tracking is not very accurate (the tracked path is larger than the actual path). I tried using the IMU values given by Realsense, but this gives a very large drift of over 10 m/s even when the camera is not moving. The IMU data given by Realsense is as follows:

Motion Intrinsic Parameters:
Motion Intrinsic of "Gyro"	  MOTION_XYZ32F
Bias Variances: 	0.000000499999999  0.000000499999999  0.000000499999999  
Noise Variances: 	0.000005148030141  0.000005148030141  0.000005148030141  
Sensitivity : 
     0.997071       0.000000       0.000000       0.002461  
     0.000000       0.997241       0.000000       0.001515  
     0.000000       0.000000       1.000773       0.000101  

Motion Intrinsic of "Accel"	  MOTION_XYZ32F
Bias Variances: 	0.000099999997474  0.000099999997474  0.000099999997474  
Noise Variances: 	0.000066952452471  0.000066952452471  0.000066952452471  
Sensitivity : 
     1.022346       0.000000       0.000000       0.027311  
     0.000000       1.029969       0.000000       0.111944  
     0.000000       0.000000       1.021683      -0.032778  

I used the above information to fill in the IMU parameters in the config file:
acc_n: 0.000066952452471        # accelerometer measurement noise standard deviation. 
gyr_n: 0.000005148030141        # gyroscope measurement noise standard deviation.     
acc_w: 0.000099999997474       # accelerometer bias random work noise standard deviation. 
gyr_w: 0.000000499999999       # gyroscope bias random work noise standard deviation

May I know if my parameters are correct? What should I use for the IMU parameters to improve the tracking performance?

mildwinter, 2019-11-12:
Is your problem solved? I also used the same problem with Microsoft's azure kinect device, which drifted a lot and looked more like imu data.

yx0123, 2019-11-14:
> Is your problem solved? I also used the same problem with Microsoft's azure kinect device, which drifted a lot and looked more like imu data.

No, I just stick with the default values given in VINS Mono

mildwinter, 2019-11-14:
Your imu parameter is a bit problematic. You need to divide acc_n and gyr_n by srqt (1/imu frequency (hz)), acc_w and gyr_w multiply by srqt (1/imu frequency (hz)) to get the discretized parameters, then I reduced the frequency of imu from 1.6khz to 50hz, and then re-improved using the imu_utils tool. The results are much better, not so drastic, but the results are not particularly good. I tried a lot of imu frequencies, 100hz and 200hz will report an error: imu message in disorder! On my device, if the frequency is higher than 500hz, the drift will be particularly serious.

canberkgurel, 2020-05-10:
@shaozu can you address this issue?

canberkgurel, 2020-05-11:
@mildwinter @yx0123 in my case these parameters were exposed in the ros topics by the ros-wrapper provided by the manufacturer.