HKUST-Aerial-Robotics/VINS-Mono #306: Vins_estimator process dies at MarginalizationFactor

minghanz, 2019-08-07:
I installed VINS-Mono on my Ubuntu16.04 laptop and successfully run EUROC's MH_01_easy.bag without a problem. 

When I shift to my own data (1080P monocular images ~20Hz, IMU ~100Hz, I adjusted the config and launch files accordingly), the [vins_estimator-2] process died after the ROS_WARN message "gyroscope bias initial calibration..." is printed, showing the following: 
[image]

Then I used backward [https://github.com/bombela/backward-cpp](url) to track the problem and got the following: 
[image]

Could anyone explain what happened and how to fix it? Thanks a lot!

zeroAska, 2019-08-13:
I have the sane issue here. Any suggestions? Thanks!

vaibhav06891, 2020-01-14:
I have run into same issue. Were either of you able to solve it ?