HKUST-Aerial-Robotics/VINS-Mono #308: Coordinate frame of the Map

oonchulip97, 2019-08-19:
I am currently using Intel RealSense T265 (with realsense-ros wrapper) with VINS-Mono.

When I started vins-estimator and moved the camera forward by 10cm, it could be seen in vins-rviz that the camera moved along the negative direction of y-axis with respect to the `world` frame. My question is, why doesn't the camera moved along the positive x-direction of the `world` frame. My assumption is based on the system initialize the `world` frame with x-direction pointing forward, y-direction pointing to the left, and z-direction pointing upwards.

Here is an image showing rviz with `World` coordinate frame.
[image]

For reference, here is the launch file and configuration file.
<launch>
    <arg name="config_path" default = "$(find feature_tracker)/../config/realsense/realsense_t265.yaml" />
	  <arg name="vins_path" default = "$(find feature_tracker)/../config/../" />
    
    <node name="feature_tracker" pkg="feature_tracker" type="feature_tracker" output="log">
        <param name="config_file" type="string" value="$(arg config_path)" />
        <param name="vins_folder" type="string" value="$(arg vins_path)" />
    </node>

    <node name="vins_estimator" pkg="vins_estimator" type="vins_estimator" output="screen">
       <param name="config_file" type="string" value="$(arg config_path)" />
       <param name="vins_folder" type="string" value="$(arg vins_path)" />
    </node>

    <node name="pose_graph" pkg="pose_graph" type="pose_graph" output="screen">
        <param name="config_file" type="string" value="$(arg config_path)" />
        <param name="visualization_shift_x" type="int" value="0" />
        <param name="visualization_shift_y" type="int" value="0" />
        <param name="skip_cnt" type="int" value="0" />
        <param name="skip_dis" type="double" value="0" />
    </node>

</launch>

%YAML:1.0

#common parameters
imu_topic: "/camera/imu"
image_topic: "/camera/fisheye1/image_raw"
output_path: "/home/parallels/Downloads/"

#camera calibration 
model_type: KANNALA_BRANDT
camera_name: camera
image_width: 848
image_height: 800
projection_parameters:
   k2: -0.0019227640004828572
   k3: 0.036574698984622955
   k4: -0.03472825139760971
   k5: 0.005598408170044422
   mu: 285.03021240234375
   mv: 286.1366882324219
   u0: 421.5474853515625
   v0: 393.88470458984375

#Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 0   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.
                        # 2  Don't know anything about extrinsic parameters. You don't need to give R,T. We will try to calibrate it. Do some rotation movement at beginning.                        
#If you choose 0 or 1, you should write down the following matrix.
#Rotation from camera frame to imu frame, imu^R_cam
extrinsicRotation: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [-0.999951, 0.00472753, 0.00865324,
          -0.00475751, -0.999983, -0.00344801, 
           0.00863679, -0.00348901, 0.999957]
#Translation from camera frame to imu frame, imu^T_cam
extrinsicTranslation: !!opencv-matrix
   rows: 3
   cols: 1
   dt: d
   data: [0.0106994807720184, 5.09053788846359e-05, -9.24136184039526e-05]
#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
equalize: 1             # if image is too dark or light, trun on equalize to find enough features
fisheye: 0             # if using fisheye, trun on it. A circle mask will be loaded to remove edge noisy points

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n:  0.08 # 0.00095       # accelerometer measurement noise standard deviation.
gyr_n:  0.004 # 0.000073         # gyroscope measurement noise standard deviation.
acc_w:  0.00004 # 0.0000071      # accelerometer bias random work noise standard deviation.
gyr_w:  2.0e-6 # 0.000000035       # gyroscope bias random work noise standard deviation.
g_norm: 9.81007     # gravity magnitude

#loop closure parameters
loop_closure: 1                    # start loop closure
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
fast_relocalization: 1             # useful in real-time and large project
pose_graph_save_path: "/home/parallels/Downloads/output/pose_graph/" # save and load path

#unsynchronization parameters
estimate_td: 0                      # online estimate time offset between camera and imu
td: 0.0                             # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#rolling shutter parameters
rolling_shutter: 0                  # 0: global shutter camera, 1: rolling shutter camera
rolling_shutter_tr: 0               # unit: s. rolling shutter read out time per frame (from data sheet). 

#visualization parameters
save_image: 0                   # save image in pose graph for visualization prupose; you can close this function by setting 0
visualize_imu_forward: 0        # output imu forward propogation to achieve low latency and high frequence results
visualize_camera_size: 0.4      # size of camera marker in RVIZ

levenberg, 2019-10-25:
@oonchulip97 Did you solve this problem? I am also running on T265. I set **rs_t265.launch** in realsense-ros with `<arg name="enable_sync"           default="true"/>` and `<arg name="unite_imu_method"      default="copy"/>`. However, when I run vins, it shows  _imu message in disorder!_. Did you have this problem?

canberkgurel, 2020-05-10:
@oonchulip97 On a separate yet related note, when I run 
`roslaunch realsense2_camera rs_t265.launch`
the available rostopics are: 
```
/camera/accel/imu_info
/camera/accel/sample
/camera/fisheye1/camera_info
/camera/fisheye1/image_raw
/camera/fisheye1/image_raw/compressed
/camera/fisheye1/image_raw/compressed/parameter_descriptions
/camera/fisheye1/image_raw/compressed/parameter_updates
/camera/fisheye1/image_raw/compressedDepth
/camera/fisheye1/image_raw/compressedDepth/parameter_descriptions
/camera/fisheye1/image_raw/compressedDepth/parameter_updates
/camera/fisheye1/image_raw/theora
/camera/fisheye1/image_raw/theora/parameter_descriptions
/camera/fisheye1/image_raw/theora/parameter_updates
/camera/fisheye2/camera_info
/camera/fisheye2/image_raw
/camera/fisheye2/image_raw/compressed
/camera/fisheye2/image_raw/compressed/parameter_descriptions
/camera/fisheye2/image_raw/compressed/parameter_updates
/camera/fisheye2/image_raw/compressedDepth
/camera/fisheye2/image_raw/compressedDepth/parameter_descriptions
/camera/fisheye2/image_raw/compressedDepth/parameter_updates
/camera/fisheye2/image_raw/theora
/camera/fisheye2/image_raw/theora/parameter_descriptions
/camera/fisheye2/image_raw/theora/parameter_updates
/camera/gyro/imu_info
/camera/gyro/sample
/camera/odom/sample
/camera/realsense2_camera_manager/bond
/camera/tracking_module/parameter_descriptions
/camera/tracking_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```
where did the `/camera/imu` in your config file come from? Is it possible to publish the 
`/camera/gyro/sample` and `/camera/accel/sample` contents on the same topic, e.g., `/camera/imu`?

canberkgurel, 2020-05-11:
@levenberg try to use `linear_interpolation` instead of `copy`.