HKUST-Aerial-Robotics/VINS-Mono #313: There is an abnormal path in VIO path when I evatuate the path precision by evo

TouchDeeper, 2019-09-20:
Hello,
I run the euroc_mh_05 dataset and write the path topic to txt file by adding the code below in the [estimator_node.cpp](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/estimator_node.cpp#L320)
```
            estimator.foutVIO <<fixed
                              << header.stamp.toSec() << " "
                              << estimator.Ps[WINDOW_SIZE].x() << " "
                              << estimator.Ps[WINDOW_SIZE].y() << " "
                              << estimator.Ps[WINDOW_SIZE].z() << " "
                              << tmp_Q.x() << " "
                              << tmp_Q.y() << " "
                              << tmp_Q.z() << " "
                              << tmp_Q.w() << endl;
```

I use the evo to evaluate the VIO path.
The trajectory aligned is as the [image](https://www.dropbox.com/s/0sj7kbrxyria4kw/2019-09-20%2013-47-13%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png?dl=0) shows.
As you can see, there is an abnormal path．
I check the output txt file and point out the corresponding abnormal path.  The abnormal path start from line 15 and end at line 47 and keep the same in each run. An example path txt file is in [here](https://www.dropbox.com/s/gnbl6pob1nwmmab/vio1.txt?dl=0).

Could you please provide some advice? Thanks a lot!

shaozu, 2019-09-20:
Hi, I test on my PC, and the result path is fine, no such problem.

TouchDeeper, 2019-09-20:
Could you please push your test code and the result path file?  I don't know what's wrong with my code. Thanks a lot!

shaozu, 2019-09-20:
The code is just the github one, and here is the [result path file](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/files/3634641/vins_result_odometry.txt). This file is generated by [those code](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/utility/visualization.cpp#L156).

TouchDeeper, 2019-09-20:
I need the path generated by pure VIO without pose graph optimization and loop closure.  

The path I need seems like is drawn by VIO plugin in rviz. The VIO plugin subscribe the topic `/vins_estimator/path`. This topic is published by these [code](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/utility/visualization.cpp#L135). 

So I just output the path as the topic` /vins_estimator/path`
```
header.stamp.toSec() 
estimator.Ps[WINDOW_SIZE].x() 
estimator.Ps[WINDOW_SIZE].y() 
estimator.Ps[WINDOW_SIZE].z() 
tmp_Q.x()
tmp_Q.y() 
tmp_Q.z() 
tmp_Q.w() 
```