HKUST-Aerial-Robotics/VINS-Mono #321: When i run roslaunch vins_estimator euroc.launch ,the error is shown:

xingxing4182, 2019-12-10:
[pose_graph-4] process has died [pid 7433, exit code -11, cmd /home/computer_823/vins_ws/devel/lib/pose_graph/pose_graph __name:=pose_graph __log:=/home/computer_823/.ros/log/373fe0f6-1b96-11ea-beb8-34f64b2c6663/pose_graph-4.log].
log file: /home/computer_823/.ros/log/373fe0f6-1b96-11ea-beb8-34f64b2c6663/pose_graph-4*.log

anyone know what is the reason about this problem? thanks very much!

hnsywangxin, 2020-04-23:
I have the same problem,my error is caused by boost conflict,my computer have two boost(1.58.0 and 1.65.0),I remove 1.65.0,it works

FrancescoFallica, 2020-04-30:
This problem showed after I modified the lines of code as suggested during the compilation:

`/home/francesco/catkin_ws/src/VINS-Mono/pose_graph/src/ThirdParty/VocabularyBinary.hpp: In constructor ‘VINSLoop::Vocabulary::Vocabulary()’:
/home/francesco/catkin_ws/src/VINS-Mono/pose_graph/src/ThirdParty/VocabularyBinary.hpp:31:11: warning: ‘VINSLoop::Vocabulary::nodes’ will be initialized after [-Wreorder]
     Node* nodes;
           ^~~~~
/home/francesco/catkin_ws/src/VINS-Mono/pose_graph/src/ThirdParty/VocabularyBinary.hpp:29:13: warning:   ‘int32_t VINSLoop::Vocabulary::nWords’ [-Wreorder]
     int32_t nWords;
             ^~~~~~
/home/francesco/catkin_ws/src/VINS-Mono/pose_graph/src/ThirdParty/VocabularyBinary.cpp:5:1: warning:   when initialized here [-Wreorder]
 VINSLoop::Vocabulary::Vocabulary()
 ^~~~~~~~`

In fact, in VocabularyBinary.hpp I moved `Node* nodes;` after `int32_t nNodes;`
The warning disappeared, but when I run the roslaunch file `[pose_graph-4] process has died...` appeared.
Set back the normal code solved my problem.