HKUST-Aerial-Robotics/VINS-Mono #331: Issue trying to run two vins-mono instances

BjarkeHellden, 2020-03-14:
I have a gazebo simulation with two robots, each with camera and IMU. I am able to use vins-mono to estimate each robot pose individually. However, when I try to run both simultaneously, only one robot is estimated correctly while the other pose estimation is garbage.

The imu and camera topics are correctly set to the corresponding robots in the config files.

My launch file looks like this:

<launch>
    <arg name="config_path1" default = "$(find feature_tracker)/../config/euroc/euroc_config1.yaml" />
	  <arg name="vins_path1" default = "$(find feature_tracker)/../config/../" />
    
    <node name="feature_tracker" pkg="feature_tracker" type="feature_tracker" output="log">
        <param name="config_file" type="string" value="$(arg config_path1)" />
        <param name="vins_folder" type="string" value="$(arg vins_path1)" />
    </node>

    <node name="vins_estimator" pkg="vins_estimator" type="vins_estimator" output="screen">
       <param name="config_file" type="string" value="$(arg config_path1)" />
       <param name="vins_folder" type="string" value="$(arg vins_path1)" />
    </node>

    <node name="pose_graph" pkg="pose_graph" type="pose_graph" output="screen">
        <param name="config_file" type="string" value="$(arg config_path1)" />
        <param name="visualization_shift_x" type="int" value="0" />
        <param name="visualization_shift_y" type="int" value="0" />
        <param name="skip_cnt" type="int" value="0" />
        <param name="skip_dis" type="double" value="0" />
    </node>

<group ns="/asd">
    <arg name="config_path2" default = "$(find feature_tracker)/../config/euroc/euroc_config2.yaml" />
	  <arg name="vins_path2" default = "$(find feature_tracker)/../config/../" />

    <node name="feature_tracker" pkg="feature_tracker" type="feature_tracker" output="log">
        <param name="config_file" type="string" value="$(arg config_path2)" />
        <param name="vins_folder" type="string" value="$(arg vins_path2)" />
    </node>

    <node name="vins_estimator" pkg="vins_estimator" type="vins_estimator" output="screen">
       <param name="config_file" type="string" value="$(arg config_path2)" />
       <param name="vins_folder" type="string" value="$(arg vins_path2)" />
    </node>

    <node name="pose_graph" pkg="pose_graph" type="pose_graph" output="screen">
        <param name="config_file" type="string" value="$(arg config_path2)" />
        <param name="visualization_shift_x" type="int" value="0" />
        <param name="visualization_shift_y" type="int" value="0" />
        <param name="skip_cnt" type="int" value="0" />
        <param name="skip_dis" type="double" value="0" />
    </node>
</group>

</launch>

What am I doing wrong and do you have any suggestions?

shaozu, 2020-03-17:
Hi, you may need to change the topic name published by `feature_tracker`.

BjarkeHellden, 2020-03-21:
Thanks for the reply.

Do you mean change the topic name from inside the feature_tracker_node.cpp ? Or using remapping ?

It seems that if I run the feature tracker node using a namespace as shown below, it will get stuck.

<launch>

<group ns="/robot1">
    <arg name="config_path1" default = "$(find feature_tracker)/../config/euroc/euroc_config1.yaml" />
	  <arg name="vins_path1" default = "$(find feature_tracker)/../config/../" />
    
    <node name="feature_tracker" pkg="feature_tracker" type="feature_tracker" output="log">
        <param name="config_file" type="string" value="$(arg config_path1)" />
        <param name="vins_folder" type="string" value="$(arg vins_path1)" />
    </node>

    <node name="vins_estimator" pkg="vins_estimator" type="vins_estimator" output="screen">
       <param name="config_file" type="string" value="$(arg config_path1)" />
       <param name="vins_folder" type="string" value="$(arg vins_path1)" />
    </node>

    <node name="pose_graph" pkg="pose_graph" type="pose_graph" output="screen">
        <param name="config_file" type="string" value="$(arg config_path1)" />
        <param name="visualization_shift_x" type="int" value="0" />
        <param name="visualization_shift_y" type="int" value="0" />
        <param name="skip_cnt" type="int" value="0" />
        <param name="skip_dis" type="double" value="0" />
    </node>
</group>

</launch>