HKUST-Aerial-Robotics/VINS-Mono #335: Some question about the implement code for IMU re-propagation

ChuangbinC, 2020-03-29:
Hi, I have a question about the implement code for IMU re-propagation.

You use the first order approximation of α,β,γ with respect to biases to  correct pre-integration results approximately instead of re-propagation in your paper (Formula 12 in the paper). But I don't find the implement code about this.  I only find the re-propagation code when the bias changes.

[image]

I want to ask if the first order approximation code is not implemented in the program?

Thanks.

shaozu, 2020-03-30:
Hi, @ChuangbinC. Check [here](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/factor/integration_base.h#L170) if you are interested.

ChuangbinC, 2020-03-30:
@shaozu Thanks for your answer. 
This method is only used in the calculation of IMU residuals. It still use re-propagation to calculate the IMU values after  Gyroscope Bias Calibration. Like [this](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/initial/initial_aligment.cpp#L35).

Is my understanding correct?
Thanks.

shaozu, 2020-03-30:
@ChuangbinC, Yes. For the code you mentioned, from scratch re-propagation is necessary because we just calibrate the gyroscope bias out, and that change may be too big to adjust according to equation (12).

 As you see, equation (12) is used in the [`evaluate` function](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/factor/integration_base.h#L176). That is, during the ceres optimization process, bias may change a little bit, and perform a from-scratch re-propagation is computational expensive.

ChuangbinC, 2020-03-30:
@shaozu Thanks for your help. I have understood this.

ChuangbinC, 2020-04-01:
@shaozu I have another question. If the bias have a little big change in the optimization process, the first order approximation  may not be approximate and it may affect the accuracy of the results. But I don't see the code to judge this situation. For example, judging whether the difference of bias is greater than a certain threshold.

shaozu, 2020-04-02:
@ChuangbinC Usually the change is small, but re-propagation according to threshold is surely doable.

ChuangbinC, 2020-04-02:
@shaozu Thanks for your answer. It helps me a lot.

closed 2020-04-02 by ChuangbinC

sviisc, 2020-04-15:
Hi, I am trying to understand pre-integration concept implemented by VINS. I have a question regarding linearized_ba and linearized_bg in [https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/factor/integration_base.h](url). I am not able to see how these two variables are getting updated. After being initialized during new instance of class, in all other functions of this class, their value look to be copied from initial value. I am also unable to see how jacobian and covariance are being used in order to update linearized bias values. Am I missing something in understanding?