HKUST-Aerial-Robotics/VINS-Mono #337: Some question about solveGyroscopeBias

ChuangbinC, 2020-04-05:
Hi, I have a question about the gyroscope bias calibration. why [this](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/initial/initial_aligment.cpp#L35) only use bgs[0] to repropagate imu data without other data like bgs[1].

ylmzkaan, 2020-04-29:
bgs is a pointer to Vector3d object. bgs[0] is the Vector3d object itself, it is not the first element of the vector. So it passes all the vector object to the function, not the first element.

EmiyaEstelle, 2020-05-08:
Hi，I got the same question as yours. Did you have something clear about it? Thank you!

qiaozhijian, 2022-01-10:
because bgs are assumed to be immutable during slide window.