HKUST-Aerial-Robotics/VINS-Mono #342: when i use my own data, i met an error: [feature_tracker-2] process has died

708949503, 2020-06-29:
As shown in after:

penCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/core/src/matrix.cpp, line 492
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/core/src/matrix.cpp:492: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

[feature_tracker-2] process has died [pid 13411, exit code -6, cmd /home/zwz/catkin_vins/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/zwz/.ros/log/4f86534e-ba00-11ea-a00e-b46921808f0e/feature_tracker-2.log].
log file: /home/zwz/.ros/log/4f86534e-ba00-11ea-a00e-b46921808f0e/feature_tracker-2*.log

can you tell me how to solve it, thank you.
i can run dataset euroc, but my own data is not work

708949503, 2020-06-29:
i have solved this question

GuoChuan1314, 2020-06-29:
How did you solve it?Can you tell me ?

708949503, 2020-06-29:
you must check your photo size 
 

> How did you solve it?Can you tell me ?

thai01255nt, 2020-07-15:
I think problem is in cv::FileStorage function (when load file config pose_graph_node) in CameraFactory.cc file. When i try include opencv2/opencv.hpp instead of opencv/core/core.hpp. tmThis function cv::FileStorage has no error. But in CameraFactort.cc source code, it include opencv/core/core.hpp. cv::FileStorage has error and terminate thread inside opencv.
I not yet fully test this. Tomorow i will test again and feedback to you.

fourteen14fourteen, 2020-08-20:
Hello, I would like to ask, are you using your own device to try to run VINS, or using your own pre-collected data set? If be oneself collect good data set, how is collect excuse me?

fourteen14fourteen, 2020-08-20:
Hello, I would like to ask, are you using your own device to try to run VINS, or using your own pre-collected data set? If be the data set that collects beforehand, how is collect excuse me?

708949503, 2020-08-24:
> Hello, I would like to ask, are you using your own device to try to run VINS, or using your own pre-collected data set? If be the data set that collects beforehand, how is collect excuse me?
You have two ways;
one is drive your own device in ROS and just need change the topic in VINS
or you collect data before hand and make rosbag bt yourself . May be the kalibr is an useful method.