HKUST-Aerial-Robotics/VINS-Mono #344: roslaunch vins_estimator euroc.launch  error

GuoChuan1314, 2020-07-01:
[feature_tracker-2] process has died [pid 98170, exit code -11, cmd /home/lab/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/lab/.ros/log/2e3d29ce-bb5e-11ea-8f87-13614794949c/feature_tracker-2.log].
log file: /home/lab/.ros/log/2e3d29ce-bb5e-11ea-8f87-13614794949c/feature_tracker-2*.log

How to solve it?Can you give me some advices?Thank  you  very much

ZhangYu-2019, 2020-07-05:
hi , I have the same problem. Have you solved it？
can you give me some advices? Thank you very much

armankuzembayev, 2020-08-14:
Try to reinstall it, Vins_mono is not working with all versions of dependent libraries (see Prerequisites in the main page). I faced the same problem. Solved by updating libraries and reinstalling

Yeguiiren, 2020-10-25:
hi, are you running this on ubuntu20.04, when i ran 'roslaunch vins_estimator euroc.launch', i have encountered the same problem, and i can't solve it. Thank you!

FSRyan, 2021-07-26:
hi,I have the same problem, did you solve it? Thank you for your reply

PuYuuu, 2021-07-26:
I have no problems using Ubuntu18.04.

FSRyan, 2021-07-26:
> I have no problems using Ubuntu18.04.

thank you! but i want to know how to solve this problem on Ubuntu 20.04

FSRyan, 2021-07-27:
I have solved the problem.This is an opencv conflict problem.
Modify the camera_model package to be compatible with opencv4.
`Chessboard.h add
#include <opencv2/imgproc/types_c.h>
#include <opencv2/calib3d/calib3d_c.h>
CameraCalibration.h add
#include <opencv2/imgproc/types_c.h>
#include <opencv2/imgproc/imgproc_c.h>`

PuYuuu, 2021-07-27:
Got it, thanks!

FGsusuCoding, 2022-06-03:
> > I have solved the problem.This is an opencv conflict problem. Modify the camera_model package to be compatible with opencv4. `Chessboard.h add #include <opencv2/imgproc/types_c.h> #include <opencv2/calib3d/calib3d_c.h> CameraCalibration.h add #include <opencv2/imgproc/types_c.h> #include <opencv2/imgproc/imgproc_c.h>`
> 
> Hi, I have the same problem in ubuntu20.04, after I tried this method(Thanks for providing this), it still hasn’t been resolved. I also tried to reinstall cv_bridge and prevent it from pointing to opencv4, but it’s still not effectiveto use. Could you please help me?thanks！ Otherwise, could you tell me the version of your eigen and ceres? I see some posts said that the problem may be a version conflict, thanks!

me too   !  did you solve it? Thank you for your reply

Eddy202208, 2022-07-28:
Hi I have a question when you install ceres-solver do you use sudo make install ? I have seen that they don't recommend to do this but if I don't do it it doesn't install ceres-solver well, any suggestions ?

LvDingzhou, 2022-09-22:
> Hi I have a question when you install ceres-solver do you use sudo make install ? I have seen that they don't recommend to do this but if I don't do it it doesn't install ceres-solver well, any suggestions ?

download ceres-solver here
[http://ceres-solver.org/installation.html](url)
click 'tgz' to install ceres-solver-1.12.0 which is the same with HKUST's
then follow the step here
[http://ceres-solver.org/installation.html](url)
and.. there won't be wrong on ceres-solver.(for me,it works)