HKUST-Aerial-Robotics/VINS-Mono #345: IMU residual at eq 16. Biases not being updated after the optimization

amirmoeini, 2020-07-03:
Hi

When we look at the IMU residual at eq 16 of the VINS-Mono paper, it seems the b_a and b_\omega are only present in that last two rows and not anywhere else and therefore during the optimisation they should be kept constant (i.e. b_{a} (k+1)-b_{a} (k+1)=0).

Can anyone please explain this?

Thanks,

[image]

Amir

shaozu, 2020-08-20:
Hi, @amirmoeini. From section IV.B.(3) of [vins-mono paper](https://ieeexplore.ieee.org/document/8421746), you may find that the first three residuals are also related to Ba and Bg.

amirmoeini, 2020-08-20:
Thanks for you reply @shaozu. 

Actually, based on my understanding of the paper, in the first three residuals, only the numerical values of \hat{\alpha}, \hat{\beta} and \hat{\gamma} are used (calculated by equation 27). So, they are not directly involved in the first three rows. So in the optimization, they will be calculated only based on the last two rows and therefore will be kept constant during optimization. 

Thanks

shaozu, 2020-08-20:
When calculate \hat{\alpha}, \hat{\beta} and \hat{\gamma} in equation 27, Ba and Bg are used. Notice that those states may change during optimization, we must apply corrections to \hat{\alpha}, \hat{\beta} and \hat{\gamma} in order to reflect the small change of Ba and Bg. In short, `Ba, Bg change` -> `\hat{\alpha}, \hat{\beta} and \hat{\gamma} change` -> `first three residuals change`.

amirmoeini, 2020-08-20:
Yes. This is clear from the paper. But this happens only two iterations and after that Ba and Bg remain constant. Based on equation 16, the solution for Ba and Bg are

b_{a} (k+1)=b_{a} (k).
b_{g} (k+1)=b_{g} (k).

So after the second iteration the Ba and Bg will remain constant.  

Please note that the **numerical values** of \hat{\alpha}, \hat{\beta} and \hat{\gamma} are used in the first three rows, and there fore they do not affect the optimization problem in solving for Ba and Bg. These two are analytically only present in the last two rows and based on that the solution for optimization is

b_{a} (k+1)=b_{a} (k).
b_{g} (k+1)=b_{g} (k).

shaozu, 2020-08-20:
> Please note that the numerical values of \hat{\alpha}, \hat{\beta} and \hat{\gamma} are used in the first three rows, and there fore they do not affect the optimization problem in solving for Ba and Bg.

As I mentioned before, \hat{\alpha}, \hat{\beta} and \hat{\gamma} DO affect the optimization through Ba and Bg correction. I recommend you to read the [corresponding code](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/factor/integration_base.h#L176), then things should be clear.

amirmoeini, 2020-08-20:
Thanks for your replies. 

Based on the corresponding code you mentioned, I assume the values for \hat{\alpha}, \hat{\beta} and \hat{\gamma} in the first three rows of the residual, are calculated by equation 33 and this actually makes sense since in this case the Ba and Bg would be present in the first three rows and involving in the optimization. However, this is not consistent with the paper as from my understanding of the paper, \hat{\alpha}, \hat{\beta} and \hat{\gamma} in the residual are calculated by equation 27. Maybe I am wrong and it is mentioned somewhere in the paper that eq 33 is used in the residual 16.

Thanks

hhhcoder, 2021-06-21:
Hi, you can see the jacobian code, if residual has no-zero jacobian on variable, variable will change during optimization.