HKUST-Aerial-Robotics/VINS-Mono #346: catkin_make error on Windows

gokberkgul, 2020-07-07:
I am trying to build VINS-Mono on Windows 10. I have installed ceres and eigen dependencies. Then, I have install ROS noetic on windows with desktop-full version. (Refer to: http://wiki.ros.org/Installation/Windows). 

ROS location: C:\opt\ros\noetic\x64
Vins-Mono location: C:\opt\ros\noetic\catkin_ws\src
Other library (like boost) locations: C:\opt\rosdeps\x64\include
Ceres Location: C:\Users\aguel\Desktop\Vins\ceres-solver (Which I point in CMakeList.txt files of camera_model, pose_estimator, vins_estimator)

I use Visual Studio 2019 Developer Command Prompt v16.6.2 and open ros bat to work on ROS environment. When I write 

`catkin_make -DCMAKE_INSTALL_PREFIX="C:\\opt\\ros\\noetic\\catkin_ws\\install"`

on the folder location "C:\opt\ros\noetic\catkin_ws" I get this error

C:\opt\ros\noetic\catkin_ws>catkin_make -DCMAKE_INSTALL_PREFIX="C:\\opt\\ros\\noetic\\catkin_ws\\install"
Base path: C:\opt\ros\noetic\catkin_ws
Source space: C:\opt\ros\noetic\catkin_ws\src
Build space: C:\opt\ros\noetic\catkin_ws\build
Devel space: C:\opt\ros\noetic\catkin_ws\devel
Install space: C:\opt\ros\noetic\catkin_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\opt\ros\noetic\catkin_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.26.28806.0
Copyright (C) Microsoft Corporation.  All rights reserved.

####
#### Running command: "nmake" in "C:\opt\ros\noetic\catkin_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.26.28806.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[  2%] Built target Project__setup_util.py_exec_install_python
[  4%] Building CXX object VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++11'
cl : Command line warning D9002 : ignoring unknown option '-O3'
cl : Command line warning D9002 : ignoring unknown option '-g'
cl : Command line warning D9002 : ignoring unknown option '-rdynamic'
benchmark_publisher_node.cpp
C:\opt\rosdeps\x64\include\boost-1_66\boost/config/stdlib/dinkumware.hpp(99): warning C4668: '_HAS_NAMESPACE' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/predef/other/endian.h(151): warning C4668: 'BOOST_ARCH_AMD64' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared\ntverp.h(119): warning C4668: 'DBG' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared\ntverp.h(126): warning C4668: 'BETA' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\shared\ntverp.h(132): warning C4668: 'OFFICIAL_BUILD' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\time.h(36): warning C4820: '_timespec64': '4' bytes padding added after data member '_timespec64::tv_nsec'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\time.h(43): warning C4820: 'timespec': '4' bytes padding added after data member 'timespec::tv_nsec'
C:\opt\rosdeps\x64\include\boost-1_66\boost/throw_exception.hpp(3): warning C4668: '__GNUC__' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/throw_exception.hpp(3): warning C4668: '__GNUC_MINOR__' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\sys/timeb.h(27): warning C4820: '__timeb32': '2' bytes padding added after data member '__timeb32::dstflag'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\sys/timeb.h(35): warning C4820: '__timeb64': '2' bytes padding added after data member '__timeb64::dstflag'
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\sys/timeb.h(44): warning C4820: 'timeb': '2' bytes padding added after data member 'timeb::dstflag'
C:\opt\rosdeps\x64\include\log4cxx/logstring.h(23): warning C4619: #pragma warning: there is no warning number '4231'
C:\opt\rosdeps\x64\include\log4cxx/logstring.h(82): warning C4668: 'LOG4CXX_CHARSET_EBCDIC' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\log4cxx/helpers/class.h(23): warning C4619: #pragma warning: there is no warning number '4231'
C:\opt\rosdeps\x64\include\log4cxx/helpers/objectptr.h(30): warning C4668: 'LOG4CXX_HELGRIND' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\log4cxx/helpers/object.h(105): warning C4625: 'log4cxx::helpers::Object::ClazzObject': copy constructor was implicitly defined as deleted
C:\opt\rosdeps\x64\include\log4cxx/helpers/object.h(105): warning C4626: 'log4cxx::helpers::Object::ClazzObject': assignment operator was implicitly defined as deleted
C:\opt\rosdeps\x64\include\log4cxx/helpers/objectimpl.h(61): warning C4820: 'log4cxx::helpers::ObjectImpl': '4' bytes padding added after data member 'log4cxx::helpers::ObjectImpl::ref'
C:\opt\rosdeps\x64\include\log4cxx/helpers/objectimpl.h(61): warning C4820: 'log4cxx::helpers::ObjectImpl': '4' bytes padding added after data member '<alignment member>'
C:\opt\rosdeps\x64\include\log4cxx/level.h(59): warning C4625: 'log4cxx::Level::LevelClass': copy constructor was implicitly defined as deleted
C:\opt\rosdeps\x64\include\log4cxx/level.h(59): warning C5026: 'log4cxx::Level::LevelClass': move constructor was implicitly defined as deleted
C:\opt\rosdeps\x64\include\log4cxx/level.h(59): warning C4626: 'log4cxx::Level::LevelClass': assignment operator was implicitly defined as deleted
C:\opt\rosdeps\x64\include\log4cxx/level.h(59): warning C5027: 'log4cxx::Level::LevelClass': move assignment operator was implicitly defined as deleted
C:\opt\rosdeps\x64\include\log4cxx/level.h(260): warning C4820: 'log4cxx::Level': '4' bytes padding added after data member 'log4cxx::Level::level'
C:\opt\rosdeps\x64\include\log4cxx/level.h(264): warning C4820: 'log4cxx::Level': '4' bytes padding added after data member 'log4cxx::Level::syslogEquivalent'
C:\opt\rosdeps\x64\include\log4cxx/level.h(264): warning C4820: 'log4cxx::Level': '4' bytes padding added after data member '<alignment member>'
C:\opt\ros\noetic\x64\include\ros/console.h(199): warning C4820: 'ros::console::FilterParams': '4' bytes padding added after data member 'ros::console::FilterParams::line'
C:\opt\ros\noetic\x64\include\ros/console.h(207): warning C4820: 'ros::console::FilterParams': '4' bytes padding added after data member 'ros::console::FilterParams::level'
C:\opt\ros\noetic\x64\include\ros/console.h(268): warning C4820: 'ros::console::LogLocation': '2' bytes padding added after data member 'ros::console::LogLocation::logger_enabled_'
[... 135 lines cut ...]
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/type_with_alignment.hpp(160): note: see reference to class template instantiation 'boost::alignment_of<boost::detail::max_align>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/is_default_constructible.hpp(16): warning C4668: 'BOOST_GCC_VERSION_WORKAROUND_GUARD' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/is_default_constructible.hpp(16): warning C4668: 'BOOST_GCC_VERSION' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/is_default_constructible.hpp(20): warning C4668: 'BOOST_GCC_VERSION_WORKAROUND_GUARD' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/is_default_constructible.hpp(20): warning C4668: 'BOOST_GCC_VERSION' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/is_default_constructible.hpp(36): warning C4668: 'BOOST_GCC_VERSION_WORKAROUND_GUARD' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
C:\opt\rosdeps\x64\include\boost-1_66\boost/type_traits/is_default_constructible.hpp(36): warning C4668: 'BOOST_GCC_VERSION' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
[... cut ...]
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed

This stackoverflow post (and other post as well) mentioned that this error is caused by "if you have boost-python in your project, boost incorrectly prevents static-linking of boost with the /MT switch.". But I couldn't see any boost python Vins-MONO project. Any idea?

shaozu, 2020-07-08:
You can try to run with our docker version.

gokberkgul, 2020-07-17:
Is it possible to access to Windows API from Docker Container? I need it in order to read sensor data and feed the program real-time.

shaozu, 2020-07-17:
You can send the sensor data to vins(docker container) via ros message.

gokberkgul, 2020-09-11:
I have built the program inside WSL, it works fine

closed 2020-09-11 by gokberkgul