HKUST-Aerial-Robotics/VINS-Mono #353: VINS-Mono Body-Of-Knowledge

AIWerkstatt, 2020-08-25:
Hi Tong Qin, Peiliang Li, and Shaojie Shen,
thank you very much for your great work on VINS-Mono and sharing your code on GitHub. After many years in medical imaging, I moved back into robotics (my dream is to build a humanoid robot). So over the last year, I have been reading up a lot and used VINS-Mono to work through the details of a state-of-the-art SLAM system. While I was working through it, I created a document that describes the **Body-Of-Knowledge behind VINS-Mono**. In the spirit of your open-software development, I thought it is only fair if I share the document. 
Check out: [http://www.aiwerkstatt.com/robotics/humanoid/vision/slam/vins-mono/ ](http://www.aiwerkstatt.com/robotics/humanoid/vision/slam/vins-mono/ )
Thank you so much,
Holger

You-zeyi, 2020-09-15:
Bravo!

lapchun, 2021-04-12:
Good job!