HKUST-Aerial-Robotics/VINS-Mono #359: Could NOT find camera_model (missing: camera_model_DIR)

zhaoworking, 2020-11-10:
`zhao@zhao-pc:~/catkin_ws/src/VINS-Mono-master$ roslaunch vins_estimator euroc.launch
... logging to /home/zhao/.ros/log/9a2cdafc-2333-11eb-a6fc-fc4596424543/roslaunch-zhao-pc-17125.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zhao-pc:33983/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/zhao/catkin...
 * /feature_tracker/vins_folder: /home/zhao/catkin...
 * /pose_graph/config_file: /home/zhao/catkin...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /vins_estimator/config_file: /home/zhao/catkin...
 * /vins_estimator/vins_folder: /home/zhao/catkin...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [17135]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9a2cdafc-2333-11eb-a6fc-fc4596424543
process[rosout-1]: started with pid [17146]
started core service [/rosout]
**ERROR: cannot launch node of type [feature_tracker/feature_tracker]: Cannot locate node of type [feature_tracker] in package [feature_tracker]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[vins_estimator-3]: started with pid [17153]
ERROR: cannot launch node of type [pose_graph/pose_graph]: Cannot locate node of type [pose_graph] in package [pose_graph]. Make sure file exists in package path and permission is set to executable (chmod +x)**
`
When I launch `vins_estimator euroc.bash` on Ubutu18.04 with Ros melodic publish version, there comes a puzzling issue that it can't launch the node of type `[feature ],[pose_graph ] `, respectively.
 So, I think there must be the problem with the `feature_tracker` and `pose_graph` which haven't been compiled.So , i go to compile them ,but unfortunately , I found that in the `CMakeList` ,the camera_model has been included, but it seems like there is not a package called `camera_model` in Ros-package list . 
What should i do , thanks in advance .

Eddy202208, 2022-07-28:
Any solutions ?

Jawaheru, 2024-03-18:
its 2024, any solution yet?

cll-git, 2024-07-05:
When I run "catkin_make" in the terminator, this bug would not happen, however it appears when I configurate the project in clion.
Any solution?
Thank you.