HKUST-Aerial-Robotics/VINS-Mono #366: MH_05_difficult.bag数据集测试出现回环检测，但轨迹输出没有更新

Andycheng0614, 2021-01-21:
在跑MH_05_difficult.bag数据集的时候，成功出现了回环检测，但组最终的轨迹没有变化，误差较大的地方仍然存在，但demo视频中误差较大的地方却明显得到了优化。

estimator终端出现如下的错误提示:
[ INFO] [1611226563.016102440]: Initialization finish!
position: 0.543290, -0.059625,optimize pose graph 
WARNING: Logging before InitGoogleLogging() is written to STDERR
E0121 10:56:22.426913 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 1.569769, 1.351370, 0.5826optimize pose graph 
E0121 10:56:24.429440 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 10.136824, 7.92optimize pose graph 
E0121 10:57:24.436502 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 8.362724, 8.290691optimize pose graph 
E0121 10:57:26.448446 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 5.447370, 9.651049, 2.740234optimize pose graph 
E0121 10:57:28.461603 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 4.281151, 10.5841optimize pose graph 
E0121 10:57:30.473723 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
optimize pose graph 
E0121 10:57:32.492225 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 4.199optimize pose graph 6884
E0121 10:57:44.508184 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 1.60274optimize pose graph 82
E0121 10:57:46.525117 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
optimize pose graph 
E0121 10:57:48.543292 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 0.386496, 0.23288optimize pose graph 
E0121 10:57:50.567509 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.
position: 0.331078, 0.271849, -0.055431

感觉是定位了回环位置，但在进行优化的时候报错了，有出现这种情况的同学么？ 该如何解决呢？

Andycheng0614, 2021-01-22:
“E0121 10:57:50.567509 17752 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.”报错是由于ceres编译安装的版本不支持EIGENSPARSE，通过"cmake -D EIGENSPARSE=ON .."重新编译安装ceres之后，就可以解决。

重新进行MH_05_difficult.bag数据集测试，出现回环检测之后，对之前的轨迹有所刷新，但优化的幅度比较小，整个数据跑完之后，还是有比较大的直观误差，与视频中的结果有比较大的差别，视频中右边的轨迹吻合的比较好，但我的实验结果还是有肉眼可见的大误差。

LIULING666666, 2024-05-27:
您好，我执行完这个cmake -D EIGENSPARSE=ON ..还是出现这种问题E0527 20:17:59.991061 131309 solver.cc:487] Terminating: Can't use SPARSE_NORMAL_CHOLESKY as sparse_linear_algebra_library_type is NO_SPARSE.，您知道怎么解决吗