HKUST-Aerial-Robotics/VINS-Mono #370: Issue in pre-integration

niebayes, 2021-02-18:
As shown in the [codes](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/factor/integration_base.h#L95), there're some expressions involve addition between rotation matrices. I think it does not make any sense to add together two rotation matrices as the rotation matrices are not in the vector space. The suggestion is adding together the quaternions and normalizing the resulted quaternion first and then transform it to the corresponding rotation matrix. A better solution might be handling the rotation directly in the manifold and then transformed back.

1ADZX, 2021-07-24:
查看你也是中国人，就不用英语了啊。感觉你上面说的很棒啊，关于VINS-mono的预积分，我还有一个疑问啊，VINS_mono中预积分中delta_V的初值为什么要设置为0，或者我观察到，只要是涉及IMU中值积分的时候，比如说这里的V,如predict()中的tmp_V,或者是processIMU()中的V[j],还有预积分中的midPointIntegration()的delta_V的初值都为0，因为这个速度值关系到两帧IMU数据之间的位置变换，感觉如果我这些速度值初值要是不为0，岂不是算出来的IMU位置就不一样了？所以我很好奇为什么都要将这个速度初值设置为0，我自己仿真的IMU数据，每一帧数据速度初值是不为0的，岂不是在这里就推算错误了？
先感谢为敬啊

1ADZX, 2021-07-24:
> 如[代码](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/dc62ec4cfdb9c420c729524c86a32de4e6759088/vins_estimator/src/factor/integration_base.h#L95)所示，有一些表达式涉及旋转矩阵之间的加法。我认为将两个旋转矩阵加在一起没有任何意义，因为旋转矩阵不在向量空间中。建议是先将四元数加在一起，然后将结果四元数归一化，然后将其转换为相应的旋转矩阵。更好的解决方案可能是直接在流形中处理旋转，然后再转换回来。
> 
> 关于如何求旋转的均值，参考可以[此处](https://www.zhihu.com/question/439497100/answer/1683258444)。

看你是中国人，就不用英语了啊。感觉你上面说的很清楚啊，关于VINS-mono的预评分，我还有一个疑问啊，VINS_mono中预评分中delta_V的初值为什么要设置为0 ，或者我观察到，只要是涉及到IMU中值积分的时候，就如这里的V，如predict()中的tmp_V，或者processIMU()中的V[j]，还有预积分中的midPointIntegration( )的delta_V的初值都为0，因为这个速度值关系到两帧IMU的位置变换，如果我这些速度值初值可能不为0，那岂不是算出来的IMU就一样了？所以我很好奇为什么要把这个速度初值设置为0，我自己仿真的IMU数据，每一帧数据速度初值是不为0的，岂不是在这里就推算错误了？
先感谢为敬啊