HKUST-Aerial-Robotics/VINS-Mono #377: VINS-Mono crashed

deeprun, 2021-03-12:
After making some code changes for OpenCV 4, I was able to build the workspace on Ubunto 18.0, ROS Melodic, OpenCV 4.0. However, when I ran the launch file

roslaunch vins_estimator euroc.launch 
It crashed:

setting /run_id to ee366926-838a-11eb-ba83-204ef6c60bc7
process[rosout-1]: started with pid [11702]
started core service [/rosout]
process[feature_tracker-2]: started with pid [11709]
process[vins_estimator-3]: started with pid [11710]
process[pose_graph-4]: started with pid [11711]
[ INFO] [1615591832.080219939]: init begins
[ INFO] [1615591832.096831964]: Loaded config_file: /home/nvidia/racecar_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
[vins_estimator-3] process has died [pid 11710, exit code -11, cmd /home/nvidia/racecar_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/nvidia/.ros/log/ee366926-838a-11eb-ba83-204ef6c60bc7/vins_estimator-3.log].
log file: /home/nvidia/.ros/log/ee366926-838a-11eb-ba83-204ef6c60bc7/vins_estimator-3*.log
[feature_tracker-2] process has died [pid 11709, exit code -11, cmd /home/nvidia/racecar_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/nvidia/.ros/log/ee366926-838a-11eb-ba83-204ef6c60bc7/feature_tracker-2.log].
log file: /home/nvidia/.ros/log/ee366926-838a-11eb-ba83-204ef6c60bc7/feature_tracker-2*.log
[pose_graph-4] process has died [pid 11711, exit code -11, cmd /home/nvidia/racecar_ws/devel/lib/pose_graph/pose_graph __name:=pose_graph __log:=/home/nvidia/.ros/log/ee366926-838a-11eb-ba83-204ef6c60bc7/pose_graph-4.log].
log file: /home/nvidia/.ros/log/ee366926-838a-11eb-ba83-204ef6c60bc7/pose_graph-4*.log

How can I fix it?

deeprun, 2021-03-13:
I found it is because of this line: 

cv::FileStorage fsSettings(config_file, cv::FileStorage::READ);

But I do not know why this could be a problem?

panmengjia, 2021-04-08:
I have the same problem.How did you solve it
 Ubunto 18.0, ROS Melodic, OpenCV 4.5