HKUST-Aerial-Robotics/VINS-Mono #378: VINS-Mono crashed on ubuntu 18.0

deeprun, 2021-03-13:
Hi, 

I am using Jetson Xavier NX, Ubuntu 18.0, ROS Melodic, Opencv 4.1. I was able to built, but could not run the vins_estimator node. It crashed at the following line:

`cv::FileStorage fsSettings(config_file, cv::FileStorage::READ);`

 I do not know if the line above has problem, or my environment setup has problem. Has anyone successfully run this repo on Ubuntu 18.0?

germanferrando, 2021-08-11:
Hi, you should install OpenCv 2.4.9. I tried with a lot of versions and that is the one, purge your OpenCV and try with this one

omeredemen, 2024-04-01:
hello @germanferrando how do you install opencv 2.4.9 on jetson nano? 
I am getting some error and i couldn't fix it

omeredemen, 2024-04-05:
After I had faced this issue i tried visual odometry with [open-vins](https://github.com/rpng/open_vins) ROS1 and i faced exactly same issue here after some research someone fixed this with installing opencv 3.4.1. It worked for me as well here is the instruction i followed for installing opencv: https://github.com/smtsarial/openvins-jetson?tab=readme-ov-file#opencv