HKUST-Aerial-Robotics/VINS-Mono #380: “roslaunch vins_estimator euroc.launch” failed in Ubuntu18

Ginger-Light, 2021-04-16:
After I input this command, the error in the output picture is output, did anyone can help me? thank u very much!

... logging to /home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/roslaunch-ginger-OptiPlex-9020-3713.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ginger-OptiPlex-9020:45857/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/ginger/catk...
 * /feature_tracker/vins_folder: /home/ginger/catk...
 * /pose_graph/config_file: /home/ginger/catk...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /vins_estimator/config_file: /home/ginger/catk...
 * /vins_estimator/vins_folder: /home/ginger/catk...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    pose_graph (pose_graph/pose_graph)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [3727]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 29e2d5a6-9e4b-11eb-a24c-3417ebd6966e
process[rosout-1]: started with pid [3738]
started core service [/rosout]
process[feature_tracker-2]: started with pid [3741]
process[vins_estimator-3]: started with pid [3742]
process[pose_graph-4]: started with pid [3743]
[ INFO] [1618533132.470710282]: init begins
[ INFO] [1618533132.500832975]: Loaded config_file: /home/ginger/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml

[vins_estimator-3] process has died [pid 3742, exit code -11, cmd /home/ginger/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/vins_estimator-3.log].
log file: /home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/vins_estimator-3*.log
[pose_graph-4] process has died [pid 3743, exit code -11, cmd /home/ginger/catkin_ws/devel/lib/pose_graph/pose_graph __name:=pose_graph __log:=/home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/pose_graph-4.log].
log file: /home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/pose_graph-4*.log
[feature_tracker-2] process has died [pid 3741, exit code -11, cmd /home/ginger/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/feature_tracker-2.log].
log file: /home/ginger/.ros/log/29e2d5a6-9e4b-11eb-a24c-3417ebd6966e/feature_tracker-2*.log

The computer configuration is: Processor: Intel Core i7-4779 CPU @3.40Hz x 8 / System:Ubuntu18.04.5 LTS

closed 2021-04-17 by Ginger-Light

resplendent-star, 2021-05-23:
hi,how do you solve this problem?

Ginger-Light, 2021-05-23:
> hi,how do you solve this problem?
hi，yes，I uninstalled my OpenCV 4.1.0，then installed a lower version 3.4.0，and the error gone

resplendent-star, 2021-05-23:
> > hi,how do you solve this problem?
> > hi，yes，I uninstalled my OpenCV 4.1.0，then installed a lower version 3.4.0，and the error gone
thanks you  for  reply
hi,when i  compile  VINS-MONO with opencv 4.1.0,i need to replace many "CV_xx_xx" to  "cv::xx_xx",so compile with opencv 3.4.0,you dont need to do that?