HKUST-Aerial-Robotics/VINS-Mono #384: problem with datatype/md5sum

sbdjyl, 2021-05-18:
my environment:
ubuntu 16.04
realsense d435i  with LibRealSense  v2.41.0

i want to use vins-mono and fast-planner in real world.
when i run vins,and then run fast-planner,i see that:
`
Client [/fast_planner_node] wants topic /vins_estimator/point_cloud to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [sensor_msgs/PointCloud/d8e9c3f5afbdd8a130fd1d2763945fca]. Dropping connection.`

It make me can't use depth image in fast-planner.
how can i solve the problem?

kalaivanank04, 2023-09-23:
i face the same issue. 
Could you please tell me the workaround?
Thanks in advance

HaiLinHou, 2024-08-15:
In vins the point cloud is in sensor_msgs/PointCloud format, but in fast-planner the point cloud is in sensor_msgs/PointCloud2 format, do a format conversion.

Conversions have ready-to-use library functions