HKUST-Aerial-Robotics/VINS-Mono #390: hi，how can i solve the problem of "Linear solver failure. Failed to compute a step: Eigen failure"?

hfw6310, 2021-10-10:
I run VINS but I also meet the question below,  can you help me and give me some advice?
[ INFO] [1633764303.358300665]: Not enough features or parallax; Move device around
[ INFO] [1633764303.558178967]: Not enough features or parallax; Move device around
[ INFO] [1633764303.758299942]: Not enough features or parallax; Move device around
[ INFO] [1633764303.958166582]: Not enough features or parallax; Move device around
[ INFO] [1633764304.158256845]: Not enough features or parallax; Move device around
[ INFO] [1633764304.358125587]: Not enough features or parallax; Move device around
[ INFO] [1633764304.558297627]: Not enough features or parallax; Move device around
[ INFO] [1633764304.758651285]: Not enough features or parallax; Move device around
[ INFO] [1633764304.958149101]: IMU excitation not enouth!
[ INFO] [1633764304.958319364]: Not enough features or parallax; Move device around
[ INFO] [1633764305.158121615]: IMU excitation not enouth!
[ INFO] [1633764305.158291854]: Not enough features or parallax; Move device around
[ INFO] [1633764305.358410640]: IMU excitation not enouth!
[ INFO] [1633764305.358723958]: Not enough features or parallax; Move device around
[ INFO] [1633764305.558236489]: IMU excitation not enouth!
[ INFO] [1633764305.558596570]: Not enough features or parallax; Move device around
[ INFO] [1633764305.758165483]: IMU excitation not enouth!
[ INFO] [1633764305.758447708]: Not enough features or parallax; Move device around
[ INFO] [1633764305.958109735]: IMU excitation not enouth!
[ INFO] [1633764305.958321996]: Not enough features or parallax; Move device around
[ INFO] [1633764306.158214085]: IMU excitation not enouth!
[ INFO] [1633764306.158411138]: Not enough features or parallax; Move device around
[ INFO] [1633764306.358250626]: IMU excitation not enouth!
[ INFO] [1633764306.358558447]: Not enough features or parallax; Move device around
[ INFO] [1633764306.558188121]: IMU excitation not enouth!
[ INFO] [1633764306.558436383]: Not enough features or parallax; Move device around
[ INFO] [1633764306.758064883]: IMU excitation not enouth!
[ INFO] [1633764306.758290994]: Not enough features or parallax; Move device around
[ INFO] [1633764306.958310455]: IMU excitation not enouth!
[ INFO] [1633764306.958641351]: Not enough features or parallax; Move device around
[ INFO] [1633764307.158264721]: IMU excitation not enouth!
[ INFO] [1633764307.158581090]: Not enough features or parallax; Move device around
[ INFO] [1633764307.358123889]: IMU excitation not enouth!
[ INFO] [1633764307.358290225]: Not enough features or parallax; Move device around
[ INFO] [1633764307.558282806]: IMU excitation not enouth!
[ INFO] [1633764307.558596312]: Not enough features or parallax; Move device around
[ INFO] [1633764307.758248169]: IMU excitation not enouth!
[ INFO] [1633764307.758574275]: Not enough features or parallax; Move device around
[ INFO] [1633764307.958218904]: IMU excitation not enouth!
[ INFO] [1633764307.958496764]: Not enough features or parallax; Move device around
[ INFO] [1633764308.158156280]: IMU excitation not enouth!
[ INFO] [1633764308.158315890]: Not enough features or parallax; Move device around
[ INFO] [1633764308.358280377]: IMU excitation not enouth!
[ INFO] [1633764308.358640134]: Not enough features or parallax; Move device around
[ INFO] [1633764308.558358099]: IMU excitation not enouth!
[ INFO] [1633764308.558651638]: Not enough features or parallax; Move device around
[ INFO] [1633764308.758152844]: IMU excitation not enouth!
[ INFO] [1633764308.758382023]: Not enough features or parallax; Move device around
[ INFO] [1633764308.958133184]: IMU excitation not enouth!
[ INFO] [1633764308.958352007]: Not enough features or parallax; Move device around
[ INFO] [1633764309.158225924]: IMU excitation not enouth!
[ INFO] [1633764309.158515069]: Not enough features or parallax; Move device around
[ INFO] [1633764309.358443489]: IMU excitation not enouth!
[ INFO] [1633764309.358728786]: Not enough features or parallax; Move device around
[ INFO] [1633764309.558129282]: IMU excitation not enouth!
[ INFO] [1633764309.558291429]: Not enough features or parallax; Move device around
[ INFO] [1633764309.758148877]: IMU excitation not enouth!
[ INFO] [1633764309.758341960]: Not enough features or parallax; Move device around
[ INFO] [1633764309.958265854]: IMU excitation not enouth!
[ INFO] [1633764309.958630636]: Not enough features or parallax; Move device around
[ INFO] [1633764310.058207336]: IMU excitation not enouth!
[ INFO] [1633764310.058542901]: Not enough features or parallax; Move device around
[ INFO] [1633764310.158208541]: IMU excitation not enouth!
[ INFO] [1633764310.158981968]: Not enough features or parallax; Move device around
[ INFO] [1633764310.258126896]: IMU excitation not enouth!
[ INFO] [1633764310.258525405]: Not enough features or parallax; Move device around
[ INFO] [1633764310.358749119]: IMU excitation not enouth!
[ INFO] [1633764310.358977601]: Not enough features or parallax; Move device around
[ INFO] [1633764310.458231506]: IMU excitation not enouth!
[... 23 lines cut ...]
[ INFO] [1633764311.658400245]: IMU excitation not enouth!
[ INFO] [1633764311.658642242]: Not enough features or parallax; Move device around
[ INFO] [1633764311.758465089]: IMU excitation not enouth!
[ INFO] [1633764311.758843903]: Not enough features or parallax; Move device around
[ INFO] [1633764311.858437011]: IMU excitation not enouth!
[ INFO] [1633764311.858713801]: Not enough features or parallax; Move device around
[ INFO] [1633764311.958231290]: IMU excitation not enouth!
WARNING: Logging before InitGoogleLogging() is written to STDERR
W1009 15:25:12.022457 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.023494 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.028426 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.029412 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.032196 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.034991 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.035960 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.042563 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.045603 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
W1009 15:25:12.048583 23662 levenberg_marquardt_strategy.cc:114] Linear solver failure. Failed to compute a step: Eigen failure. Unable to perform dense Cholesky factorization.
[ INFO] [1633764312.058313866]: IMU excitation not enouth!
[ INFO] [1633764312.158444527]: IMU excitation not enouth!
[ INFO] [1633764312.258694541]: IMU excitation not enouth!
[ INFO] [1633764312.258930884]: Not enough features or parallax; Move device around

[image]
@qintonguav

EriqZhao, 2022-10-09:
Hello. I recently experienced the same problem, as well as a serious shift in the trajectory during the initial phase of SLAM. Do you know how to solve this problem?

YoungHaulk, 2023-02-19:
> 

Hello, I got  the same question. How  did you  solve it?

WFram, 2023-02-20:
There might be several reasons why those log messages occur:

1) IMU data stream issues. Check device hardware connection and IMU data frequency. Also, check if the units of IMU readings are m/s^2 for the accelerometer data and rad/s for the gyro data;
2) Camera and IMU unsynchronization. If a time offset between two data streams is too large, VINS cannot perform.
3) Extrinsic between camera and IMU. Check if the relative transformation between two sensors is calibrated correctly.

In addition, VINS needs some movement of a device to initialize inertial variables. Also, make sure you perform enough rotation and translation of the device and the camera does not look at a flat surface.

YoungHaulk, 2023-02-23:
> There might be several reasons why those log messages occur:
>
> 1. IMU data stream issues. Check device hardware connection and IMU data frequency. Also, check if the units of IMU readings are m/s^2 [...]

Thank you very much for your answer！
The reprojection error result of the Extrinsic between the camera and the imu is 0.6 pixels (the calibration tool used is kalibr).So I don't think it is caused by external parameter calibration.
Before the image is input into vins, I have done some preprocessing operations. Therefore, I suspect that there is a problem with the time stamp synchronization of the image and the imu.
Thank you, brother！

mansisarawata, 2024-09-27:
@hfw6310 were you able to solve this?

mansisarawata, 2024-09-27:
@WFram   I am running into the same errors. Basically while using VINS-Mono and VINS-Fusion for pinhole mono-imu setup it throws out error message: “Not enough features or parallax; Move device around” even when there is enough motion and different features in the environment. The camera data and imu data are collected from iphone 12 and published via two separate nodes and the timestamps are matched by rospy.Time.now() and the frequency of imu is 100hz and that of camera is 30fps. Camera data is retrieved from a video file and imu from csv. Parameters are similar to euroc dataset’s mono-inertial config and VINS-Mono/Fusion is run by setting 2 in config ie we have no prior idea of the extrinsics. The rest of the parameters are adjusted to camera specifications(ie frame height,width etc). Can you help me out on this? Like what might be causing this issue?

lida2003, 2024-11-03:
Have you anyone get this solved? I have the same issue. My time delay for image an imu is kind of close to rosbag file.

Anything else might cause this?

[image]