HKUST-Aerial-Robotics/VINS-Mono #400: Vins Mono crashed on Nvidia Jetson Nano

Avi241, 2022-02-08:
Hello Everyone I have succescfully compiled Vins mono on Nvidia jetson nano but when I run 
`roslaunch vins_estimator euroc.launch` 
the program crashes.

> [ INFO] [1644341394.759498831]: init begins
[ INFO] [1644341394.776419664]: Loaded config_file: /home/arvind/catkin_ws/src/VINS-Mono/feature_tracker/../config/euroc/euroc_config.yaml
[vins_estimator-3] process has died [pid 3197, exit code -11, cmd /home/arvind/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/arvind/.ros/log/b8f60f66-8904-11ec-b2c5-845cf3270371/vins_estimator-3.log].
log file: /home/arvind/.ros/log/b8f60f66-8904-11ec-b2c5-845cf3270371/vins_estimator-3*.log
[feature_tracker-2] process has died [pid 3188, exit code -11, cmd /home/arvind/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/arvind/.ros/log/b8f60f66-8904-11ec-b2c5-845cf3270371/feature_tracker-2.log].
log file: /home/arvind/.ros/log/b8f60f66-8904-11ec-b2c5-845cf3270371/feature_tracker-2*.log
[pose_graph-4] process has died [pid 3198, exit code -11, cmd /home/arvind/catkin_ws/devel/lib/pose_graph/pose_graph __name:=pose_graph __log:=/home/arvind/.ros/log/b8f60f66-8904-11ec-b2c5-845cf3270371/pose_graph-4.log].
log file: /home/arvind/.ros/log/b8f60f66-8904-11ec-b2c5-845cf3270371/pose_graph-4*.log

According to this issue [#377 ] 

`cv::FileStorage fsSettings(config_file, cv::FileStorage::READ);`

this is the reason for crash.

Can anyone help me with this ?

didzdissanayaka8, 2022-02-08:
Hi, Did you do any changes to the code or it is just the config file you have changed? If it is only the config file, can you post it here as well?

Avi241, 2022-02-08:
No, I didn't change anything, not even a config file I have just done catkin_make.

Arghyajony, 2022-02-24:
I got similar problem while using it with jetson nano. Did you find any solution??

ArghyaChatterjee, 2022-02-27:
This is a problem with the config file not loading properly. Please check the permission and then execute.

Avi241, 2023-07-13:
Thanks for your reply @ArghyaChatterjee  but I see no problem with the config file. The problem is in the code of feature_tracker I debugged the code and the crash is happening at line 91 in feature_tracker.cpp in line
cv::Ptr<cv::CLAHE> clahe = cv::createCLAHE(3.0, cv::Size(8, 8)); . The program is crashing at this line as soon as the first image is received. Can anyone please help me with this ?? OpenCV version I am using is 4.2.0

kukulinxin, 2023-11-29:
Hello, have you solved this problem?

omeredemen, 2024-04-05:
@kukulinxin i fixed the problem. check [here](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/378#issuecomment-2039131357)

Avi241, 2024-04-05:
Thanks @omeredemen  will check it out

vitcamm, 2025-09-30:
Hello, have anyone solved this problem?