HKUST-Aerial-Robotics/VINS-Mono #408: 利用docker搭建环境，不能链接到https://ceres-solver.googlesource.com/ceres-solver/

klxxxx, 2022-06-29:
报错信息如下：
Cloning into 'ceres-solver'...
fatal: unable to access 'https://ceres-solver.googlesource.com/ceres-solver/': Failed to connect to ceres-solver.googlesource.com port 443: Connection timed out

klxxxx, 2022-06-29:
修改dockerfile中对应命令为： git clone https://github.com/ceres-solver/ceres-solver.git && \ ，重新build还是报错Cloning into 'ceres-solver'...
fatal: unable to access 'https://ceres-solver.googlesource.com/ceres-solver/': Failed to connect to ceres-solver.googlesource.com port 443: Connection timed out
，好像没有修改一样，有大神知道为什么么

af-doom, 2022-07-28:
> 修改dockerfile中对应命令为： git clone https://github.com/ceres-solver/ceres-solver.git && \ ，重新build还是报错Cloning into 'ceres-solver'... fatal: unable to access [...]

兄弟，你解决问题了吗

xuhao1, 2022-08-17:
@klxxxx @af-doom 这是GFW引起的。

closed 2022-08-17 by xuhao1

ZJJ2022415, 2022-11-01:
请问解决了吗？

TheSeanParker, 2023-12-16:
一开始,我以为需要科学上网,后来发现,这个网站就进不去https://ceres-solver.googlesource.com/ceres-solver/