HKUST-Aerial-Robotics/VINS-Mono #417: When I use ZED2,I found an error .[feature_tracker-2]process has died [pid 7505 ,exit code -6]

MissTen, 2022-09-26:
The error like this:
OpenCV Error: Assertion failed (_mask.empty() || (_mask.type() == CV_8UC1 && _mask.sameSize(_image))) in goodFeaturesToTrack, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/imgproc/src/featureselect.cpp, line 366
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/imgproc/src/featureselect.cpp:366: error: (-215) _mask.empty() || (_mask.type() == CV_8UC1 && _mask.sameSize(_image)) in function goodFeaturesToTrack

[feature_tracker-2] process has died [pid 20370, exit code -6, cmd /home/ztt/mono/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/ztt/.ros/log/1adff86a-3d3f-11ed-a8bb-d83bbff7b92b/feature_tracker-2.log].
log file: /home/ztt/.ros/log/1adff86a-3d3f-11ed-a8bb-d83bbff7b92b/feature_tracker-2*.log

In fact,I have browsed the similar error in the issuses,but I still get confused if the error was raised by the photo size.
At the same time,I can run the euroc and the car successfully.
@

lilpeng-xxxl, 2022-10-18:
I met the same error.
Did you solve it?

MissTen, 2023-03-09:
> yes.I found that I set the size of the image to half the size of the camera，it will success,but I still can not understand why.Else,the next pace ,I can not watch the track of camera on the right image.

Badrivishal, 2023-03-10:
What do you mean by half the size of camera? I am setting it to the size specified in the Image topic(640, 360) for 720p (Which is incidentally half because each camera captures half the 720 pixels). This removes the Assertion Failed but feature tracker still crashes after a couple of frames.