HKUST-Aerial-Robotics/VINS-Mono #420: Met an Error

lilpeng-xxxl, 2022-10-18:
[image]
I met an error.When I use my own dataset,this error appeared.

lilpeng-xxxl, 2022-10-19:
I have solved it by changing the frame size to 1280x720.
Now I meet a new problem.
There is no locus on the rviz.
"Not enough features or parallax;Move your device","unstable features tracking,please slowly move your device"

keyohan, 2023-03-03:
How did you change the frame size?
What is the frame size?
My image size is 1280 x 960, i dont know how to change it. Is it fixed?

lilpeng-xxxl, 2023-03-09:
> 你是如何改变框架大小的？ 什么是框架尺寸？ 我的图像尺寸是 1280 x 960，我不知道如何更改它。它是固定的吗？

You can change the size in the yaml file. You can try 1280 x 720, that is what I set to be.

Badrivishal, 2023-03-10:
You have to change it in the .yaml file in the config folder.

mansisarawata, 2024-09-20:
> I have solved it by changing the frame size. Now I meet a new problem. There is no locus on the rviz. "Not enough features or parallax;Move your device","unstable features tracking,please slowly move your device"

@lilpeng-xxxl were you able to solve this issue using the same data?