HKUST-Aerial-Robotics/VINS-Mono #443: Cannot launch  vins_estimator euroc.launch

Jawaheru, 2024-03-18:
ERROR: cannot launch node of type [feature_tracker/feature_tracker]: Cannot locate node of type [feature_tracker] in package [feature_tracker]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [vins_estimator/vins_estimator]: Cannot locate node of type [vins_estimator] in package [vins_estimator]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [pose_graph/pose_graph]: Cannot locate node of type [pose_graph] in package [pose_graph]. Make sure file exists in package path and permission is set to executable (chmod +x)

Please note that these pkgs are in my rospack list
and i have give the access permission to all of them using chmod +x 
I use ubuntu 18.04 and ros melodic,
Help me

vipulgarg1998, 2024-03-23:
I am using ubuntu 20.04 and noetic and it ran in the first attempt. Did it build successfully on `catkin_make`?

Jawaheru, 2024-03-24:
> I am using ubuntu 20.04 and noetic and it ran in the first attempt. Did it build successfully on `catkin_make`?

Yes ofc I did