HKUST-Aerial-Robotics/VINS-Mono #444: Has anyone had succeed with rolling shutter camera and unsync low frequency IMU (MPU9250)

omeredemen, 2024-03-21:
Hi all,
I am using see3cam cu27 kamera and mpu9250. I know that the lowest robust setup is rolling shutter camera and unsync low frequency IMU but when i tried with my setup vins-mono couldn't even initialize.
[image]

omeredemen, 2024-03-22:
Well, after some calibrations I got some good calibration result. Now It can be initialized but drift occurs. I think this issue is because of imu-cam sync and bad imu timestamp. here is my result:

[image]

How can i correct the imu timestamp with ros or if there is any alternative could you please let me know?

SerpentWindy, 2024-04-24:
Also, may I ask if you frequently experience "system reboot"(Trajectory returns to origin), and if your dynamic calibration results match those of offline calibration.

omeredemen, 2024-04-29:
oh sorry for the late reply. i will try what you said about parameters. 
i actually didn't understand what you asked so can you explain if you don't mind

SerpentWindy, 2024-04-30:
> oh sorry for the late reply. i will try what you said about parameters. i actually didn't understand what you asked so can you explain if you don't mind

I explained in advance that there is no reliable evidence for the first point, only as a suggestion.
After reviewing your calibration file, I found three possible issues:

1. Your image size is (640,480), and VINS uses two feature detection parameters when processing images of (640,480) size, "min_dist" is 30 and 25, respectively. Therefore, I suggest you try using "min_dist" parameters of 30 and 25 respectively.

2. Your timeshift_cam_imu is 241ms, and according to the VINS development team, the timeshift_cam_imu should be controlled within 30ms, otherwise it will cause motion mismatch between the image and IMU data. Therefore, it is necessary to reduce the time error to within 30ms. My suggestion is that the best approach is hard synchronization. If hard synchronization is not possible, you can inquire and research the time from when the camera and IMU received the trigger signal to when you stamped it, and subtract the corresponding time from your camera timestamp and IMU timestamp. Reduce time synchronization error through the above methods.

3. Your gyroscope noise standard deviation is 0.1495249516640554, which is larger than the noise standard deviation of the accelerometer 0.05166104637481483. In my experience, this is not reasonable. I suggest that you first confirm whether there is any abnormal data in your IMU data, whether the format of the calibration data is correct (timestamp, wx, wy, wz, ax, ay, az), and check whether the unit of the IMU output angular velocity you use is deg/s or rad/s (the unit required by VINS is rad/s), and then re calibrate the IMU internal parameters. If the output angular velocity unit of your IMU gyroscope is deg/s, please multiply the IMU output angular velocity by PI/180 to convert it to rad/s. May I ask what is the IMU calibration tool you are using? If it is imu_utils, because the units of the imu_utils result are not actually rad/s and m/s ^ 2, but rad/s/sqrt (Hz) and m/s ^ 2/sqrt (Hz), you also need to multiply the noise standard deviation of the accelerometer and gyroscope by the sqrt your IMU output frequency. The random walk of the accelerometer and gyroscope is divided by the sqrt your IMU output frequency. For example, if your IMU frequency is 100Hz, then the sqrt your IMU output frequency is 10.

omeredemen, 2024-05-06:
i am using kalibr as calibrarion tool and to calculate allan variance parameters i am using [this tool](https://github.com/ori-drs/allan_variance_ros). And this repo gives allan variance parameters as kalibr requested format.

well with hardware i have now, i cant make a hard sync but i succesfully reduce the timeshift_cam_imu. But still my gyroscope and accelerameter noise standart deviation too big. I checked my acc and gyro units again and i think there is no problem with them

SerpentWindy, 2024-05-06:
> i am using kalibr as calibrarion tool and to calculate allan variance parameters i am using [this tool](https://github.com/ori-drs/allan_variance_ros). And this repo gives allan variance parameters as kalibr requested format.
> 
> well with hardware i have now, i cant make a hard sync but i succesfully reduce the timeshift_cam_imu. But still my gyroscope and accelerameter noise standart deviation too big. I checked my acc and gyro units again and i think there is no problem with them
> 
> I will use another mpu9250 module and calculate allan variance parameters again

Sorry, I have used the imu_utils tool instead of the one you used, so I cannot provide any feedback on this.

I have to apologize for the third point I made in my previous comment. I discussed with friends today and noticed that the noise of the gyroscope does not necessarily have to be smaller than the noise of the accelerometer, which means your calibration data may be correct.

May I ask if VINS often experiences  "system reboot" (Trajectory returns to origin) after reducing the timeshift_cam_imu?

omeredemen, 2024-05-06:
> May I ask if VINS often experiences "system reboot" (Trajectory returns to origin) after reducing the timeshift_cam_imu?

I changed my camera to raspberry pi v2 a long time ago and i dont remember I've seen this expression before sorry.
with rpi camera and mpu9250 it cannot be even initialized now i think i should change my hardware to a reliable one.
What does that mean and where does it occur from

> Sorry, I have used the imu_utils tool instead of the one you used, so I cannot provide any feedback on this.

Thanks for the clues I think this issue is because of imu-cam unsync, imu noises and camera. i tried to solve imu noises with avarege filter, but i think i should use better camera for this.

SerpentWindy, 2024-05-07:
> I changed my camera to raspberry pi v2 a long time ago and i dont remember I've seen this expression before sorry. with rpi camera and mpu9250 it cannot be even initialized now i think i should [...]

The "system reboot” is like the following figure, after running VINS for a period of time, the trajectory will jump to the starting point and run again.
[image]

> Thanks for the clues I think this issue is because of imu-cam unsync, imu noises and camera. i tried to solve imu noises with avarege filter, but i think i should use better camera for this.

I think your idea is right

SerpentWindy, 2024-05-07:
system reboot figure 
[image]

SerpentWindy, 2024-05-07:
[image]

omeredemen, 2024-05-08:
Ohh i see. Yes it was happened continuously as the following image.

> ![Screenshot from 2024-03-22 14-10-38](https://private-user-images.githubusercontent.com/102593810/315990180-f7aa9263-8194-4c31-b7c1-6d5ef6600368.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUxNDYzMjksIm5iZiI6MTcxNTE0NjAyOSwicGF0aCI6Ii8xMDI1OTM4MTAvMzE1OTkwMTgwLWY3YWE5MjYzLTgxOTQtNGMzMS1iN2MxLTZkNWVmNjYwMDM2OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwOFQwNTI3MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNGNlZGQ4M2U2NDgyZmY2M2Y0YmI5NTY1YmVlYWM1Nzg5ZmE1MTI3NmY4N2FiMTkyNjU2MTNkNWUzMWY3NWJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.rJ0gYi2Yb-hbnxdLDltnTpGu0Wwx3vegvUPsRyfqDFE)

omeredemen, 2024-05-10:
[image]