HKUST-Aerial-Robotics/VINS-Mono #446: ROS2 compatible

serkanMzlm, 2024-04-02:
Will it be made compatible with ROS2?

serkanMzlm, 2024-04-04:
As I converted the code to ROS 2, I conducted experiments with the 'V1_01_easy' bag file, and the output turned out to be different from ROS 1. Upon inspecting the output data, I noticed that the data published from the feature_tracker part remains the same, while the position data published from the VINS estimator part appears to be significantly larger, despite the orientation data being similar.

ROS:
-  /imu0 hz: 200
-  /cam0/image_raw: 19.9

ROS2:
-  /imu0 hz: 134
-  /cam0/image_raw: 19.9

[image]

rviz2:
[image]

YasheerahYaqoot, 2024-07-04:
> As I converted the code to ROS 2, I conducted experiments with the 'V1_01_easy' bag file, and the output turned out to be different from ROS 1. Upon inspecting the output data, I noticed that the data published from the feature_tracker part remains the same, while the position data published from the VINS estimator part appears to be significantly larger, despite the orientation data being similar.
> 
> ROS:
> 
> * /imu0 hz: 200
> * /cam0/image_raw hz: 19.9
> 
> ROS2:
> 
> * /imu0 hz: 134
> * /cam0/image_raw hz: 19.9
> 
> ![image](https://private-user-images.githubusercontent.com/74418302/319496389-f640516f-efb8-40c9-a169-d8ac47b7f230.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxMzA3ODcsIm5iZiI6MTcyMDEzMDQ4NywicGF0aCI6Ii83NDQxODMwMi8zMTk0OTYzODktZjY0MDUxNmYtZWZiOC00MGM5LWExNjktZDhhYzQ3YjdmMjMwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA0VDIyMDEyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjZjI1OGNhZjFkMGMzNGZjYmQ0ZDg1YjQxYmNlZjc5NmQ4NmQzZjcyMTQ2ODkwZDA2YWZjNmQwYmFlZDFiMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wJk5I4WF1u1Sj0cCSYo-jQQBKpUAP766MadWNUS4uE4)
> 
> rviz2: ![image](https://private-user-images.githubusercontent.com/74418302/319497212-dc767a4c-353f-4b16-945c-55a590a4e60d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAxMzA3ODcsIm5iZiI6MTcyMDEzMDQ4NywicGF0aCI6Ii83NDQxODMwMi8zMTk0OTcyMTItZGM3NjdhNGMtMzUzZi00YjE2LTk0NWMtNTVhNTkwYTRlNjBkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA0VDIyMDEyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlZThiMjJhN2IxYzQxNTQ5YzFlNDcxZmE1N2I1M2YxYWJhYzEyNjk5OTM0ZGQyZDdhNTlmOTZjMmY4N2RhNDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zjJwEdT1m4gJJEnwFoLIHheolGPrJmUMurvh1pJut1I)

Hello, I wanted to ask that you converted it for which ubuntu and ros2 version? Was it ubuntu 20.04 and ros2 foxy or some other?

YasheerahYaqoot, 2024-07-04:
@serkanMzlm I wanted to ask that you converted it for which ubuntu and ros2 versions? Was it for ubuntu 20.04 and ros2 foxy or some other?

serkanMzlm, 2024-07-05:
@YasheerahYaqoot  
System I Use:
Ubuntu: 22.04 
ROS2: Humble

YasheerahYaqoot, 2024-07-05:
@serkanMzlm Thank you!.
Also, can you tell me what libraries did you install? Ceres 2.1.0? Did you install opencv with CUDA or Eigen 3.3.9?

serkanMzlm, 2024-07-05:
@YasheerahYaqoot Ceres, opencv and Eigen are installed, but I don't know their versions, but it is most likely the latest version.

YasheerahYaqoot, 2024-07-05:
@serkanMzlm I m having problem in installing opencv. It gives error when i run this command : make -j8. Can you tell me how did you install it?

serkanMzlm, 2024-07-09:
@YasheerahYaqoot Why are you trying to install from source?
Can you post a screenshot of the error?
After all the requirements are met, the packages need to be built using `colcon build`
```
sudo apt-get install -y libopencv-dev
```

DMUZhangXianglong, 2024-07-09:
> @YasheerahYaqoot System: Ubuntu: 22.04 ROS2: Humble

I also want to convert this code to ROS2 hubmble in ubuntu22.04，I don‘’t know if we can discuss with each other. and my e-mail:347913076@qq.com

YasheerahYaqoot, 2024-07-10:
@serkanMzlm I did install every package and when I build the packages it gives me error. I removed everything for now. Will post a screenshot after doing it again. I m cloning your repo and installing all the dependencies and libraries and then building the package.

YasheerahYaqoot, 2024-07-10:
@DMUZhangXianglong Yes that would be great! I will email you my process and my problems and then we can try doing it together.

YasheerahYaqoot, 2024-07-10:
@serkanMzlm I get this error 
[image]

serkanMzlm, 2024-07-10:
@YasheerahYaqoot Your error is due to performing the build sequence incorrectly.

serkanMzlm, 2024-07-10:
@YasheerahYaqoot

YasheerahYaqoot, 2024-07-10:
@serkanMzlm Yes please, that would be very helpful. I think the issue is in the building of packages. I have tried several times but I am unable to do it. Let me know when you are available.

serkanMzlm, 2024-07-10:
> @serkanMzlmBu hatayı alıyorum ![görüntü](https://private-user-images.githubusercontent.com/140265389/347390917-2cfdc8ec-4898-4379-9e4f-198f8e57b3ab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA2MjA1MDYsIm5iZiI6MTcyMDYyMDIwNiwicGF0aCI6Ii8xNDAyNjUzODkvMzQ3MzkwOTE3LTJjZmRjOGVjLTQ4OTgtNDM3OS05ZTRmLTE5OGY4ZTU3YjNhYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxMFQxNDAzMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZDY0Y2Q0MGUwYWYwNjQ1OTNkYzZmMmMwMDU3MTRjOTY0YjA0ZDU2Nzk5NmI4ZjUyYWE3ZTM3Mzk1YTJmOGIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.E8FUEnxGLvFHg_gZl4PD6jJ6m1J_3naUqOF0h0L4KyU)

The cause of this error is the build order. First, please build the **camodocal** package.

YasheerahYaqoot, 2024-07-10:
I get this error when I run the colcon build for camodocal 
[image]

YasheerahYaqoot, 2024-07-10:
Maybe on Sunday you can help me with the necessary libraries and correct installing, building and sourcing since I m very new to this.

YasheerahYaqoot, 2024-07-10:
Ok so, I was able to build camodocal. And now when I finally build the worskpace it gives me error. 
[image]

serkanMzlm, 2024-07-10:
@YasheerahYaqoot Installation steps: 
1- Download the ROS2 branch on the GitHub page
```
git clone git@github.com:serkanMzlm/VINS-Mono.git -b ros2
```
2- 
```
```

YasheerahYaqoot, 2024-07-10:
I am trying to do it on orangepi 5. The package is built but it does not launch rviz.

serkanMzlm, 2024-07-10:
@YasheerahYaqoot The RViz part is still for ROS1; I haven't adapted it for ROS2. Running it manually and selecting the necessary packages should be sufficient.

DMUZhangXianglong, 2024-07-11:
> @DMUZhangXianglong Yes that would be great! I will email you my process and my problems and then we can try doing it together.

Is it possible to use instant messaging software for communication, such as WeChat

DMUZhangXianglong, 2024-07-11:
> @YasheerahYaqoot and @DMUZhangXianglong , if you want, I can show you the build part on Sunday. Then, we can work together to resolve the errors causing issues in the system.

Is it possible to use instant messaging software for communication, such as WeChat

YasheerahYaqoot, 2024-07-11:
@serkanMzlm I have a simple USB camera. But first I want to test it on the existing dataset. What should I run to know that the package is installed correctly and working fine.

YasheerahYaqoot, 2024-07-11:
@DMUZhangXianglong I am available on Telegram.

serkanMzlm, 2024-07-11:
@YasheerahYaqoot The reason for opening this topic is that I converted all ROS1 code to ROS2 and ensured it builds without errors. However, there is an issue with one package. I run the code, and it works without any problems, but the resulting mapping is not the same as the graph in ROS1. I opened this topic so that the developers can review the ROS2 code and identify the error.

DMUZhangXianglong, 2024-07-12:
> @DMUZhangXianglong I am available on Telegram.
I can only use wechat