HKUST-Aerial-Robotics/VINS-Mono #450: Can it be deployed on bare linux (not ROS)?

lida2003, 2024-05-17:
Hi I have RaspberryPi3B+ and Jetson Orin board. 
And my [initial flight with companion computer](https://www.bilibili.com/video/BV1c1421z7kz/) is OK.

I hope to run VINS-Mono on bare linux (NOT ROS). Is there any chance or  steps that I can follow?

Any link or idea is well appreciated.

SerpentWindy, 2024-05-20:
Yes, VINS can run on bare Linux. You only need to provide a middleware or data transmission architecture to replace the functionality of ROS in VINS.

lida2003, 2024-05-20:
> only need to provide a middleware or data transmission architecture

So VINS is highly intergrated with this middleware or data transmission architecture, which is currently called ROS, am I right?

SerpentWindy, 2024-05-21:
> > only need to provide a middleware or data transmission architecture
> 
> So VINS is highly intergrated with this middleware or data transmission architecture, which is currently called ROS, am I right?

yes

closed 2024-05-29 by lida2003